RedApparat/Fotoapparat #109: How to take live preview frames?

Rehckover, 2017-10-01:
I am trying to recive frames as bitmap in live preview. Do you have such callback? For example in camera api 2 its called on  public void onImageAvailable(ImageReader reader)

dmitry-zaitsev, 2017-10-02:
To do that you need to use `FrameProcessor` which you can specify while initializing `Fotoapparat` instance. Note, however, that images which you will receive will be in NV21 format, so you will need to convert them to `Bitmap` yourself.

Or just copy&paste one of StackOverflow solutions (we all do that from time to time 😉 ): https://stackoverflow.com/a/32793908/926907

closed 2017-10-02 by dmitry-zaitsev

friendoye, 2017-10-02:
Maybe add class ```FrameProcessorUtil``` with convert method and add to examples sample with using this method? So every user of this library will not copy&paste code from SO. 🤔

dmitry-zaitsev, 2017-10-03:
Converting preview frames to Bitmap is usually not a best thing to do. That is ok for proof-of-concept implementation, but if you are seriois about processing those images consider using OpenCV