RedApparat/Fotoapparat #111: Question about dynamic positioning/sizing of CameraView/TextureView

machty, 2017-10-02:
I'm refactoring some old front-facing camera code to use Fotoapparat, and I'm running across a familiar tricky case and was wondering if Fotoapparat might have a solution/API to address.

Here's the deal: I need to position the camera view right below a header with a fixed / known-up-front height, but there's this cat-and-mouse where I need to know the final resolved height that Fotoapparat is going to use for the preview size so that I can properly size the CameraView's height.

Here's a quick graphic representation of what I'm trying to do, if it helps at all:

[image]

I feel like a solution to #107 might help my situation. Basically, I don't think Fotoapparat exposes which preview size is ultimately going to be used by Android, either before or after the cameraView is displayed, which makes it difficult (without megahax), to simply align the top of the video feed with some other element in the UI.

In the meantime, I'm installing a timer that check for when the TextureView is initialized/sized so that I can dynamically adjust the height before doing `cameraView.setVisibility(View.VISIBLE)`.

Diolor, 2017-10-03:
Problem is that camera 1 only informs about available sizes when you start the camera itself.

An alternative way could be to get the available resolutions for that device with `Fotoapparat#getCapabilities()` without starting the preview (passing an empty `CameraRenderer` into the `.into(cameraView)`). 
Then you or the user decides which one of those you will render, stop the previous session and open the camera again with the normal view.

I have not tried this to know if Android Camera needs a surface to start or an exception will occur. I guess should be fine. Let us know.

Diolor, 2017-12-08:
I will close this since there is no reply. Please let me know if this need to be reopened for any reason

closed 2017-12-08 by Diolor