RedApparat/Fotoapparat #113: is there any feature to show live camera preview?

developer--, 2017-10-08:
I want to add some filters in preview, like gamma and etc

andrewcking, 2017-10-12:
You should be able to do this with the .frameProcessor(myFrameProcessor) argument when creating your fotoapparat instance, or are you looking to do something else?

developer--, 2017-10-13:
do you have any example of how to process this frame? I couldn't create bitmap from this frame, tried to get bytes from frame and than converting it to bitmap, but it returns null

fbarrosodev, 2017-10-19:
The bitmap is null because the image is in NV21 format.
Check this to convert the byte[] to bitmap.
https://stackoverflow.com/a/32793908/4148930

sksar, 2017-10-29:
This library can help,
[https://github.com/silvaren/easyrs](https://github.com/silvaren/easyrs)

Diolor, 2017-12-08:
Editing a photo/video/surface/preview stream is another topic and nothing this library can do :) Hope you found your solution!

closed 2017-12-08 by Diolor