RedApparat/Fotoapparat #114: Get relative rotation of camera to device

matej-svejda, 2017-10-09:
Im using your (awesome) library to capture images. I need to make sure that the captured images are upright (meaning that the width of the resulting, rotated image is smaller than the height) when the device is in portraite mode. However, I wasn't able to find a way to determine what the rotation of the camera sensor is in relation to the device (i.e. Camera.CameraInfo.orientation for Camera1 API). For most devices this is 90 degrees, but there is no guarantee. I need this information to correctly interpret the width and height of the supported output formats to chose the correct one.

Thanks in advance :)

Diolor, 2017-12-08:
I missed this issue. How about `Photo#rotation`?

matej-svejda, 2017-12-11:
I would need this information before capturing images. AFAIK there is no guarantee that the camera in a device is always mounted in portrait mode so if the preview says its 640x480 i dont know if it is 640 pixels in height or in width, after correct rotation. For example for wide-angle lenses it happens that in portraite mode the resulting images have higher width than height.

Diolor, 2018-01-18:
`Photo#rotation` will provide from next release on provide a rotation relative to the device orientation and not the screen mode rotation. 

fixed in #180

closed 2018-01-18 by Diolor