RedApparat/Fotoapparat #115: Camera DisplaySurface does not reinitialize onStart for Some Devices

andrewcking, 2017-10-11:
## You Guys Rock
First off, I am so grateful for Fotoapparat, it is an amazing library with excellent functionality and has been wonderful to work with. 

## The Issue
I've discovered that on some devices the camera view does not reinitialize (shows a white or black screen instead) when returning to the camera activity but it always initializes correctly the first time. 

This can be reproduced with the Fotoapparat sample app in Android Studio with the emulator running a Nexus 5 on Android 5.1.1. If you load the sample app, hit the home button and then re-enter the app the camera view will be blank.

Testing with different versions of Fotoapparat from 1.4.1 back through 1.1 yields the same result (I did not test back farther). 

The relevant logcat entries are

```
D/Fotoapparat: setDisplaySurface
D/Camera: app passed NULL surface
D/Fotoapparat: startPreview
```

## Confirmed Affected Devices

The devices I have confirmed are affected so far (non-emulated) are the LGv20, Galaxy S6 and Galaxy S5.

## Some Workarounds

I tried my hand at finding the issue but was not able to pin it down precisely. I did discover some (bad) workarounds that may offer some insight.

### Sleep before calling camera.setDisplaySurface(textureView);

If you sleep the thread temporarily in textureRendererView before calling setDisplaySurface the issue does not occur:

```
TextureRendererView.java

    @Override
    public void attachCamera(CameraDevice camera) {
        try {
            awaitSurfaceTexture();
            updateLayout(camera);
            try{
                Thread.sleep(50);
            }catch (Exception e){
                // Do nothing
            }
            camera.setDisplaySurface(textureView);
        } catch (InterruptedException e) {
            // Do nothing
        }
    }
```
Furthermore adding in a small thread sleep anywhere in the StartCameraRoutine will also fix is the issue.

### Wait to call fotoapparatSwitcher.start on onStart
On the user side the issue can be relieved by waiting a few ms before calling fotoapparatSwitcher.start() in onStart. For instance in onStart:

```
    Handler handler = new Handler();
    handler.postDelayed(new Runnable() {
        @Override
        public void run() {
            fotoapparatSwitcher.start();
        }
    }, 60);
```

### setContentView in onStart instead of onCreate

Another thing I tried that fixed the issue is to move setContentView from onCreate to onStart (along with the other dependent bits).  Obviously not optimal but it is interesting. I am not sure if this is because it forces a reinit onStart or if it works because it effectively is delaying the call to fotoapparatSwitcher.start.

```
protected void onStart() {
        super.onStart();

        setContentView(R.layout.activity_main);
        cameraView = findViewById(R.id.camera_view);
```

## Final Thoughts

Thank you again for your great library. Hopefully my odd tests provide some insight. If there is anything further I can do please let me know. This issue seems similar to #77 but since it is reported fixed I am probably wrong.

mickod, 2017-10-16:
I have also seen this issue on a Nokia 5 and a Samsung galaxy S5. 

In my case, the workaround I tried from the set above did not work - the third option above, setting the content view in onStart. 

This may be because I also have some other asynchronous work running, including location updates which may affect the timing also if the root problem is a race condition.

Any insights anyone has would be much appreciated...

mickod, 2017-10-16:
Note I think this issue raised previously is probably the same root problem - #104

andrewcking, 2017-10-16:
Interesting - that would suggest that none of the workarounds are completely safe (I am hoping someone with more experience than I, can use this info to find a real fix). I am curious to know, if you try the second workaround, 'Wait to call fotoapparatSwitcher.start on onStart' does it work? If not, if you try increasing the delay from 60ms to 400ms or something a lot bigger does it work? If so that would further suggest that the root problem is a race condition.

Crysis21, 2017-10-17:
I'm having the same issues when dealing with multiple switched between cameras and app switching. At some point, the screen remains white and if I check the cameraswitcher, it's state is initialised.

mickodo, 2017-12-08:
Ni - Any update on this issue (and on - #104)

Diolor, 2017-12-24:
You may try `2.0.0-beta2` to check if it's still happening

andrewcking, 2018-01-01:
I ran a few tests with the 2.0.0 release on the Galaxy s6 (which was previously affected) and the issue appears to be resolved (the camera displaySurface always reinitialized)! In a couple days I will test it out on the s5 and v20 and report back if anything comes up, but I am guessing this release did the trick. I cannot thank you all enough for this great library!

Diolor, 2018-01-01:
That's great news Andrew! I will mark this as resolved and if persists, feel free to comment again. Have a happy coding year!

closed 2018-01-01 by Diolor

dranzerashi, 2019-10-09:
Im having similar issue, white screen on lenovo tab 8 plus (android 9)