RedApparat/Fotoapparat #123: Consecutive photos after switch camera.

marcusfiuza, 2017-10-18:
Tryng to take consecutive photos after switch camera.

1. take a photo.
2. switch camera.

#### initialize FA
```java
return Fotoapparat
                .with(this)
                .cameraProvider(CameraProviders.v1()) // change this to v2 to test Camera2 API
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .photoSize(standardRatio(biggestSize()))
                .lensPosition(lensPosition(position))
                .focusMode(firstAvailable(
                        continuousFocus(),
                        autoFocus(),
                        fixed()
                ))
                .flash(firstAvailable(
                        autoRedEye(),
                        autoFlash(),
                        torch(),
                        off()
                ))
                .previewFpsRange(rangeWithHighestFps())
                .sensorSensitivity(highestSensorSensitivity())
                .frameProcessor(new SampleFrameProcessor())
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new CameraErrorCallback() {
                    @Override
                    public void onError(CameraException e) {
                        Toast.makeText(LabCoresSelfieActivity.this, e.toString(), Toast.LENGTH_LONG).show();
                    }
                })
                .build();

```

#### What was the result you received?
10-18 11:56:32.944 7310-7485/semprepresente.dev D/Fotoapparat: getPreviewStream
10-18 11:56:34.111 7310-7485/semprepresente.dev D/Fotoapparat: takePicture
10-18 11:56:34.729 7310-7485/semprepresente.dev D/Fotoapparat: startPreview
10-18 11:56:38.378 7310-7485/semprepresente.dev D/Fotoapparat: stopPreview
10-18 11:56:38.445 7310-7485/semprepresente.dev D/Fotoapparat: close
10-18 11:56:39.174 7310-7485/semprepresente.dev D/Fotoapparat: open
10-18 11:56:41.177 7310-7485/semprepresente.dev D/Fotoapparat: getPreviewStream
10-18 11:56:41.693 7310-7485/semprepresente.dev E/AndroidRuntime: FATAL EXCEPTION: pool-14-thread-1
                                                                             Process: semprepresente.dev, PID: 7310
                                                                             Theme: themes:{default=overlay:system, iconPack:system, fontPkg:system, com.android.systemui=overlay:system, com.android.systemui.navbar=overlay:system}
                                                                             java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                 at android.hardware.Camera.native_getParameters(Native Method)
                                                                                 at android.hardware.Camera.getParameters(Camera.java:1979)
                                                                                 at io.fotoapparat.hardware.v1.PreviewStream1.addFrameToBuffer(PreviewStream1.java:46)
                                                                                 at io.fotoapparat.hardware.v1.PreviewStream1.start(PreviewStream1.java:88)
                                                                                 at io.fotoapparat.routine.ConfigurePreviewStreamRoutine.run(ConfigurePreviewStreamRoutine.java:30)
                                                                                 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                                 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                                 at java.lang.Thread.run(Thread.java:818)

#### What did you expect?
continue to use

#### Context:

* *FA version*: 1.4.1
* *Devices/APIs affected*: motox
* *any other relevant information*:

marcusfiuza, 2017-10-23:
Any idea of how can i go arround that?

Diolor, 2017-12-31:
Should be ok in v2. Feel free to comment if it persists

closed 2017-12-31 by Diolor