RedApparat/Fotoapparat #124: Portrait photos width and height resolution are inverted

AlessandroIu, 2017-10-20:
## The issue

Photos made in portrait mode and in landscape mode has the same width and height resolution.
In my case, both have a width of 4640 pixels and height of 3480...and this is not correct for portrait mode.

## What was the result you received?

Also if I took the two following pictures after device rotation occurred, their width and height are the same.

Portrait photo
![Portrait photo]
(https://raw.githubusercontent.com/AlessandroIu/photos/master/photos/4fotoapparat/_IMAGE_20171020_134242_.jpg)

Landscape photo
![Landscape photo]
(https://raw.githubusercontent.com/AlessandroIu/photos/master/photos/4fotoapparat/_IMAGE_20171020_134255_.jpg)

Screenshot of the portrait photo with its properties
![Portrait image]
(https://raw.githubusercontent.com/AlessandroIu/photos/master/photos/4fotoapparat/Portrait.png)

Screenshot of the landscape photo with its properties
![Landscape image]
(https://raw.githubusercontent.com/AlessandroIu/photos/master/photos/4fotoapparat/Landscape.png)

Furthermore, some desktop applications and OS previewing icons software are able to correct the error, other doesn't.
For example:
- GIMP prompt the user before opening the photo telling that resolution is not correct and suggests to rotate the image;
- Ubuntu shows the image preview without problems;
- Windows shows the image preview without problems;
- Debian 9.1 previews of the image don't build properly, resulting in a corrupted image, as you can see in the following picture (pictures previewed are other pictures made by the app using Fotoapparat library).

Previews from Debian 9.1
![Previews from Debian 9.1]
(https://raw.githubusercontent.com/AlessandroIu/photos/master/photos/4fotoapparat/previews.png)

I think that orientation and resolution don't match.

## The code for taking a picture

```java

private void takePicture() {
        PhotoResult photoResult = fotoapparatSwitcher.getCurrentFotoapparat().takePicture();
                imageFile = "new_file"();

        }

        photoResult.saveToFile(imageFile);

        photoResult
                .toBitmap(scaled(0.25f))
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() [{](url)
                    @Override
                    public void onResult(BitmapPhoto result) {

                        // Previewing the photo made
                        ImageView imageView = (ImageView) findViewById(R.id.result);
                        imageView.setImageBitmap(result.bitmap);
                        imageView.setRotation(-result.rotationDegrees);
                    }
                });

    }

```

## What did you expect?
I expect to know if there is a way to manage the width and height resolution and the orientation of the photo taken, being able to match the two and ensuring that the photo is displayed unequivocally and with the correct orientation on most of the devices/pcs.
I would also like to know if there is a page/file/documentation which explains all the methods of this library...or if all the methods are already described in the "README.md" and there is no other sources as documentation for features/functionalities/methods of this library.

## Context:

* *FA version: 1.4.1
* *Devices/APIs affected: One Plus Three

**Thank you for your efforts and your time**

AlessandroIu, 2017-11-03:
I noticed that in the EXIF (containing metadata) of the photo saved, the information about the rotation of the camera at the photo shoot moment is contained.
Evidently, this information is not processed by some previews software (like Debian 9.1) while is correctly previewed by other software (Windows, Ubuntu).
I find out this by uploading the portrait image on an online EXIF reader service (like http://www.verexif.com/en/ver.php); the  "Orientation" parameter has the value  "rotate 90" .

closed 2017-11-03 by AlessandroIu

dmitry-zaitsev, 2017-11-03:
Sorry for no reply, I somehow forgot about this issue.

That's right, we write EXIF to the image file so that image in actual file is not really rotated and should be rotated by the applications. The reason is simple - it would be just too slow and memory consuming to rotate the image on a mobile device.