RedApparat/Fotoapparat #129: It is possible to have a call when the file writing on the storage is completed?

AlessandroIu, 2017-11-03:
I'm trying to wait for the completion of file saving in order to start, in another activity, the functionality for making the preview of the photo taken, and also for calling the media scanner and updating the Android gallery.
For now, the method "whenAvailable" only allows me to made a preview in the same activity (with the following code) but not to preview the image in another activity (and that's why I think that the photo is not completed when I try to preview it).

```java
    private void takePicture() {
        PhotoResult photoResult = fotoapparatSwitcher.getCurrentFotoapparat().takePicture();

        // Storing file, file name and location into the corresponding strings
                String timeStamp = new SimpleDateFormat("yyyyMMdd_HHmmss").format(new Date());
                String imageFileName =  "IMAGE_" + timeStamp + "_" + ".jpg";
                imageFile = new File(storageDirectory,imageFileName);

        // Saving the picture taken in the file
        photoResult.saveToFile(imageFile);

        // This part manages what to do when photo saving is finished
        // For now manages the preview in the CameraActivity and the media scanning of the file
        photoResult
                .toBitmap(scaled(0.25f))
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {

                        // Previewing the photo made
                        ImageView imageView = (ImageView) findViewById(R.id.result);
                        // Creating a separate task that will made the preview disappear after 5 seconds
                        new RemovingPreview(imageView, result).execute();

                    }
                });
    }
```

I expect to obtain a call to a method or something else when photo file saving is finished.

#### Context:

*FA version: 1.4.1
*Devices/APIs affected: One Plus Three

**Thank you for your time**

dmitry-zaitsev, 2017-11-03:
`saveToFile` returns `PendingResult` which also have `whenDone` callback. You can use it to know when saving is done.

closed 2017-11-03 by dmitry-zaitsev

AlessandroIu, 2017-11-03:
I've implemented the callback as (I think) you suggested me but it seems not working.
the `whenDone` callback runs before the `whenAvailable` one.
Maybe I'm missing something.
I'm attaching a screenshot made during debugging in which you can see, by the logs, that line 304 (inside the `whenDone` callback) is executed before line 316 (inside the `whenAvailable` callback)

```java

        // Saving the picture taken in the file and capturing the return value of the callback inside pr
        PendingResult<Void> pr = photoResult.saveToFile(imageFile);
        // In the following line we will wait for image file to be completely written
        pr.whenDone(new PendingResult.Callback<Void>() {
            @Override
            public void onResult(Void aVoid) {

                // Tell the MediaScanner about the new file so that it is immediately available to the
                // user via the gallery
                new SingleMediaScanner(getApplicationContext(), imageFile);
                Log.i(TAG_THIS_CLASS, "photoresult.PendingResult " + " WHEN DONE callback run");
            }
        });

        // This part manages what to do when photo saving is finished
        // For now manages the preview in the CameraActivity and the media scanning of the file
        photoResult
                .toBitmap(scaled(0.25f))
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {

                        Log.i(TAG_THIS_CLASS, "photoresult.PendingResult " + " WHEN AVAILABLE callback run");

                        // Previewing the photo made
                        ImageView imageView = (ImageView) findViewById(R.id.result);
                        // Creating a separate task that will made the preview disappear after 5 seconds
                        new RemovingPreview(imageView, result).execute();

                    }
                });
```

I used logs instead of breakpoints beause if I use the latter, the app will crash, in general, the app crash if I put breakpoints on methods of the Fotoapparat library...is this normal?

Than you again for your time.

[image]

dmitry-zaitsev, 2017-11-03:
There is no guarantee which callback will be called first - the operations are running in parallel. So, you should fix your code accordingly so that it does not depend on the exact order of the callbacks.

AlessandroIu, 2017-11-07:
Sorry, but I did not understand.
I thought that using 'whendone' I was able to activate the callback in the time when the saving process is ended (and not before).
Can I please ask you to be more explicit or giving me an example of how to do that or help me to understand what I'm missing?
Thank you again.

fyzs-n9001, 2018-04-15:
@dmitry-zaitsev  Don't play dumb , I think you know what AlessandroIu means. No matter 'whenDone' or 'whenAvailable'  can be helpful to make sure the image file is saved