RedApparat/Fotoapparat #135: Failed to open camera with...

JPLITPeople, 2017-11-24:
Hi guys,

I've been seing a lot of "errors" on the CameraErrorCallback handler:

Failed to open camera with lens position null and id:0
Failed to open camera with lens position BACK and id:0
Failed to open camera with lens position null and id:-1
Failed to start preview for camera devices:0

I don't think it crashes though as i'm not seing any crash related to this issue in my logs.

Does it just show black screen? I can't replicate it in any of my devices

friendoye, 2017-11-24:
Sounds like very familiar error on Nexus 5. Running several apps that using camera is causing apps to crash.
Try to close every app, that is using camera.

JPLITPeople, 2017-11-24:
I've tested on a nexus 5 and 5X just now and didn't got any error. Altough I just had my app and the native camera running.

Diolor, 2017-12-08:
Unfortunately it's something you need to take care e.g. display error to your user. We have the errorcallback where you can observe these issues. This happens when we try to open the camera and the hardware just denies us

closed 2017-12-08 by Diolor