RedApparat/Fotoapparat #136: orientation data in exif is always ORIENTATION_UNDEFINED

AviranAbady, 2017-11-26:
#### What are you trying to achieve or the steps to reproduce?

Read orientation data from exif

#### How did you initialize FA?

```java
// init camera
        fotoApparat = Fotoapparat
                .with(getContext())
                .cameraErrorCallback(this)
                .cameraProvider(CameraProviders.defaultProvider(getContext()))
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .build();
    }

// start camera
 fotoApparat.start();

// take picture

 fotoApparat.takePicture().toPendingResult().whenAvailable(new PendingResult.Callback<Photo>() {
                @Override
                public void onResult(Photo photo) {
                    handlePhoto(photo.encodedImage);
                }
            });
        }

// read exif from the byte array (using exif support library)
        InputStream in = null;
        ExifInterface exifInterface = null;
        try {
            in = new ByteArrayInputStream(encodedImage);
            exifInterface= new ExifInterface(in);
        }
        catch (IOException e) {}
     
        }

        exifInterface.getAttribute(ExifInterface.TAG_ORIENTATION); // <- always zero
       // ORIENTATION_UNDEFINED = 0;

```

#### What was the result you received?

Orientation data in exif is always ORIENTATION_UNDEFINED = 0;
checked on several devices

#### What did you expect?
correct orientation data

#### Context:

The activity is locked into portrait mode (using manifest)
I do rotate the device into landscape mode to take photos Photo.rotationDegrees will have correct values around 90 or 270 when device is in landscape.

FA Version: 1.4.1
Nexus 6p, Nexus 5x

AviranAbady, 2017-11-26:
I've created a demo project showcasing the issue
https://github.com/AviranAbady/FotoApparatIssueDemo

the project below has the activity set to portrait only.

The "take photo" button will  appear only when the phone is rotated to landscape position (There's an orientation listener)

Once an image is captured, exif is extracted - orientation will be undefined.

@Diolor @dmitry-zaitsev please assist, thanks.

Diolor, 2017-12-08:
Honestly I don't know how EXIF data behave. Did you find your solution?

AviranAbady, 2017-12-08:
@Diolor unfortunately, no.

Diolor, 2017-12-08:
will `photo#rotation` work for you?

AviranAbady, 2017-12-08:
@Diolor no, i cant tell from tham if the photo is upside down.

on most phones, rotating to the left (90 deg counter clock wise) will result in a normal photo
but on some phone (like nexus 5x)  if you rotate to the left (90 deg couter clock wise) you will get an upside down photo.

`photo#rotation` would be the same on both phone.

I've created a demo project, linked in the original post, if you wanna have a look.

Diolor, 2017-12-08:
https://github.com/Fotoapparat/Fotoapparat/blob/master/sample/src/main/java/io/fotoapparat/sample/MainActivity.java#L225

AviranAbady, 2017-12-08:
@Diolor it's not about how to rotate the bitmap, this is about knowing when it needs rotation.

In an app, when the activity is set to portrait only. If a photo is taken when the phone is rotated to landscape... the rotation of the image is unpredictible.

on most phones, if you rotate clock wise it will be upside down.
on some phones, it's the other way around...

currently it's impossible to know when you need to rotate it back to normal.

dmitry-zaitsev, 2018-01-18:
@AviranAbady sorry for the very late reply.

I checked your sample app. The thing is - we pass the JPEG image from the camera as is, and in our configuration the camera is not writing EXIF rotation into the image. So we are writing it manually into the file later on.

What does it mean for you:
- You can either save files using `PhotoResult.saveToFile`. In that case we write EXIF to the file.
- You can use rotation provided by `Photo` object. To see how it is converted to EXIF rotation check source code of `ExifWriter` (it is pretty small).

closed 2018-01-18 by dmitry-zaitsev

indraAsLesmana, 2018-09-21:
for everyone who reads this discussion, here's the answer:
`val orientation = ExifInterface(file.absolutePath).getAttribute(ExifInterface.TAG_ORIENTATION)`

jrocha, 2019-07-05:
I was having this issue as well. After checking checking the code I found that saveToFile write this attribute async, so I just wait it to be done.

```
photoResult.saveToFile(targetFile)
                    .whenDone(unit -> {
						// do you your stuff here
                    });
```