RedApparat/Fotoapparat #137: Include width & height of the taken image in PhotoResult

vanniktech, 2017-11-26:
It would be cool if the width & height of the image would be available in the PhotoResult class.

Diolor, 2017-11-26:
I see. How about the EXIF data of the photo? Are they empty?

vanniktech, 2017-11-26:
Can I get them easily without creating anything?

Diolor, 2017-11-26:
If you use `BitmapPhoto` you can use the [bitmap's api method](https://developer.android.com/reference/android/graphics/Bitmap.html#getWidth()).
If from a file check [step 3](https://stackoverflow.com/a/20094093/1447885) with [ExifInterface.TAG_IMAGE_WIDTH](https://developer.android.com/reference/android/media/ExifInterface.html#TAG_IMAGE_WIDTH).

Will that work?

dmitry-zaitsev, 2017-11-26:
Alternatively you can use BitmapFactory to just decode the bounds of the image (specifically by using `BitmapFactory.Options.inJustDecodeBounds`. This is a pretty fast operation.

We will consider adding the image size to the result in the future.

vanniktech, 2017-11-26:
The Exif solution would probably work. However that step, parsing the file could be saved altogether since Fotoapparat should know the size of the image already when the picture was taken. Also it feels like a good fit for the PhotoResult class.

Diolor, 2017-11-26:
[FA doesn't know the dimens](https://github.com/Fotoapparat/Fotoapparat/blob/7fbf4c1e5a26de14084b8c0bbc0947e855fc7c52/fotoapparat/src/main/java/io/fotoapparat/hardware/v1/Camera1.java#L291). 

Agree that `Photo` *could* also have the dimens

vanniktech, 2017-11-27:
Oh okay didn't know that. Then I'll leave it up to you whether you want to add support for that or not. Using ExifInterface on the consuming side is fine too.

commit cd6cebdb (2018-01-03): Photo has helping functionality to get the dimensions. Fixes #137
closed 2018-01-03 by Diolor