RedApparat/Fotoapparat #139: Camera 1 dark preview

JPLITPeople, 2017-11-28:
Some devices are showing a very dark preview comparing to stock camera (specially on low light environments).

Aparently the problem is from the devices an not Fotoapparat. The issue seems to be the Fps range reported. The problematic devices show the same number for lowest and highest range. (ex: [15000,15000] while choosing rangeWithHighestFps or rangeWithLowestFps.

The solution for me was to add a new function:

    public static SelectorFunction<Collection<Range<Integer>>, Range<Integer>> customFpsRange() {
        return new SelectorFunction<Collection<Range<Integer>>, Range<Integer>>() {
            @Override
            public Range<Integer> select(Collection<Range<Integer>> items) {
                if (items.isEmpty()) {
                    return null;
                }

                ArrayList<Range<Integer>> rangeArraylist= new ArrayList<>();
                 for (Range<Integer> range : items) {

                    if (range.highest().intValue() != range.lowest().intValue()) {

                        rangeArraylist.add(range);
                    }

                }

                if (!rangeArraylist.isEmpty()) {

                    return Collections.max(rangeArraylist, COMPARATOR_BY_BOUNDS);
                } else {

                    return Collections.max(items, COMPARATOR_BY_BOUNDS);
                }

            }
        };
    }

This gets all the ranges that have different lowest and highest values, and chooses the max from it.

This fixes most my problems with it. Altough some devices were still a bit dark, so i had to play a bit with the sensorSensitivitySelectors. Giving the lowest value would still make some devices a bit dark, but nothing compared to before

Diolor, 2017-11-28:
This is very interesting @JPLITPeople ! Do you have some example devices which have an obvious (easily reproducible) issue? 

I know that in Nexus 6P e.g. facebook messenger produces darker images than the stock camera app.

JPLITPeople, 2017-11-28:
I've tested on:
 - Nvidia Shield K1
 - Wiko pulp
 - Lenovo Phab 2 pro

But you can test it on a dark or really low light environment, switching from fotoapparat camera to stock camera and check if the image is darker. Altough not all devices will be really dark, you can spot the diferences when the "bug" occurs.

I'm using ISO around 400 as well

Diolor, 2017-12-08:
Brain dump/todo: maybe we can improve our fps selectors

Diolor, 2017-12-09:
Ok this is a bug in our selectors. Non-fixed rates must be selected earlier than fixed ones

sefaaycicek, 2017-12-29:
We are experiencing this error in our project. device is samsung grand prime.
When will you fix this problem?

Diolor, 2017-12-29:
Try v2.0.0-beta2

sefaaycicek, 2017-12-29:
Thanks. 

Try v2.0.0-beta2 doesn't open camera and gives the following error.

 

io.fotoapparat.exception.camera.CameraException: Failed to open camera with lens position: io.fotoapparat.characteristic.LensPosition$Back@ba70187 and id: 0
                                                                              at io.fotoapparat.hardware.CameraDevice.open(CameraDevice.kt:65)
                                                                              at io.fotoapparat.routine.camera.StartRoutineKt.start(StartRoutine.kt:38)
                                                                              at io.fotoapparat.routine.camera.StartRoutineKt.bootStart(StartRoutine.kt:20)
                                                                              at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:86)
                                                                              at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:40)
                                                                              at io.fotoapparat.hardware.ExecutorKt$sam$Runnable$76f0e8a1.run(Executor.kt)
                                                                              at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                              at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                              at java.lang.Thread.run(Thread.java:818)
                                                                           Caused by: java.lang.IllegalArgumentException: FocusMode mode manual is not supported.
                                                                              at io.fotoapparat.parameter.camera.convert.FocusModeConverterKt.toFocusMode(FocusModeConverter.kt:22)
                                                                              at io.fotoapparat.capability.provide.CapabilitiesProviderKt$getCapabilities$2.invoke(CapabilitiesProvider.kt:17)
                                                                              at io.fotoapparat.capability.provide.CapabilitiesProviderKt$getCapabilities$2.invoke(CapabilitiesProvider.kt)
                                                                              at io.fotoapparat.capability.provide.CapabilitiesProviderKt.extract(CapabilitiesProvider.kt:28)
                                                                              at io.fotoapparat.capability.provide.CapabilitiesProviderKt.getCapabilities(CapabilitiesProvider.kt:17)
                                                                              at io.fotoapparat.capability.provide.CapabilitiesProviderKt.getCapabilities(CapabilitiesProvider.kt:11)
                                                                              at io.fotoapparat.hardware.CameraDevice.open(CameraDevice.kt:62)
                                                                              at io.fotoapparat.routine.camera.StartRoutineKt.start(StartRoutine.kt:38) 
                                                                              at io.fotoapparat.routine.camera.StartRoutineKt.bootStart(StartRoutine.kt:20) 
                                                                              at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:86) 
                                                                              at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:40) 
                                                                              at io.fotoapparat.hardware.ExecutorKt$sam$Runnable$76f0e8a1.run(Executor.kt) 
                                                                              at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
                                                                              at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
                                                                              at java.lang.Thread.run(Thread.java:818)

Diolor, 2017-12-29:
Please keep this issue clean. This is unrelated and not and  normal exception.

You have wrong configuration and the exception is explicit enough: FocusMode mode manual is not supported.

Diolor, 2017-12-31:
New selectors added in v2

closed 2017-12-31 by Diolor