RedApparat/Fotoapparat #144: Disable confirmation after take photo

core2root, 2017-12-06:
hi , is there are way to disable "use photo" & "retake" option and immediately return the result?

dmitry-zaitsev, 2017-12-06:
Sorry, but it is not clear what your question is. Could you please provide more details?

I will close the issue until new details are available.

closed 2017-12-06 by dmitry-zaitsev

core2root, 2017-12-06:
hi, i have other issue i cant start the preview, it shows nothing.
i set the builder as in your tutorial 

fotoapparat = Fotoapparat.with(this)
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .lensPosition(LensPositionSelectors.back())
                .previewSize(SizeSelectors.biggestSize())
                .photoSize(SizeSelectors.biggestSize())
                .build();
@Override
    protected void onStart() {
        super.onStart();

        fotoapparat.start();
    }

dmitry-zaitsev, 2017-12-06:
Make sure that you have the camera permission