RedApparat/Fotoapparat #146: Running app and app is crashing in starting

GrishmaPromact, 2017-12-07:
I am running this app and app is crashed in starting.
Please do needful.
I have attached screenshot of crash log.
[image]

Diolor, 2017-12-07:
Will be fixed by #145

GrishmaPromact, 2017-12-07:
@Diolor ohk. Thanks.

GrishmaPromact, 2017-12-11:
@Diolor Still crashing is occurred. When your PR is getting merged?

GrishmaPromact, 2017-12-20:
@Diolor when it going to resolved? We have higher priority to add this library in project but because of this issue, we are not able to add that.

Diolor, 2017-12-20:
This will be fixed in 2.0. You can use `2.0.0-beta1`, if you want to, which has the fix. Be warry: apis and camera behavior might change. See `v2` branch for more details and how-to configure v2 FA.

GrishmaPromact, 2017-12-20:
v2 update - App crashed while change the mobile to portrait to landscape continuously .
I attached the screen shot of crashed log.
[image]

Diolor, 2017-12-24:
This has been fixed in `2.0.0-beta1` and I cannot reproduce it further on a pixel 2 XL

GrishmaPromact, 2017-12-26:
I have used Samsung galaxy on 5 (Crashed while change the orientation continuously) and also Moto e4 plus (In this camera is not displaying only blank white screen is displaying and crashed while i am zooming camera).

closed 2017-12-31 by Diolor