RedApparat/Fotoapparat #148: Manual focus triggers flash

Raenar4k, 2017-12-08:
#### What are you trying to achieve or the steps to reproduce?
I've implemented focus on tap, using focus or autofocus methods and flash toggle.
If flash is on, then camera will activate flash while focusing and then deactivate it, just like when it is taking a photo. 
I expect it only to focus, without turning on flash.

```kotlin
fotoapparat.focus().whenAvailable {
            Timber.tag(TAG).d("focus state is $it")
        }
```

#### How did you initialize FA?

```kotlin
        return Fotoapparat
                .with(context)
                .into(renderer)
                .previewScaleType(ScaleType.CENTER_CROP)
                .photoSize(AspectRatioSelectors.wideRatio(SizeSelectors.biggestSize()))
                .lensPosition(LensPositionSelectors.lensPosition(LensPosition.BACK))
                .focusMode(Selectors.firstAvailable(
                        FocusModeSelectors.autoFocus(),
                        FocusModeSelectors.fixed()
                ))
                .flash(FlashSelectors.off())
                .logger(Loggers.loggers(Logger { Timber.tag(CameraImpl.TAG).d(it) }))
                .cameraErrorCallback({ errorSubject.onNext(it) })
                .build()
```

#### What was the result you received?

When camera is focusing, flash is activated. 

#### What did you expect?

When camera is focusing, flash is not activated, like on default google camera.

#### Context:

* *FA version*: 1.5.0
* *Devices/APIs affected*: Seems to be not device specific (Tested on  Nexus 5x, Samsung A5, Xiaomi redmi 4, Samsung S5)
* *any other relevant information*: 

Thanks for working on this lib! :)

Diolor, 2017-12-08:
Thanks! I have seen few things with camera1 & flash so far...
Related : #99

Diolor, 2018-01-08:
I now understood what you want. No, this is not possible with android camera1 which this lib depends on. 

Reason is focus & capture photos are actions while flash state is the "how" which the former actions use.

You may try to "silently" (w/o flash) focus by having flash off, then focus, then turn flash to auto, & then (probably) take a picture with auto flash. Be wary that this lib is not made for this kind of usage and the above might result in weird behaviour because of threading issues.

Raenar4k, 2018-01-10:
I've implemented workaround similar to what you say - triggering focus sets flash selectors to OFF (if it is not disabled already), then focus is called and when focus result is delivered through callback - then flash selector is being set back to whatever setting it was before focusing.

Im not sure if that is very specific use case, since triggering focus is something fairly common for users to do. But i do understand that it could be complicated problem. 

In fact, i've seen different focus behavior in default cameras - Nexus 5x **will not** trigger flash while focusing, but Samsung A5 **will**.