RedApparat/Fotoapparat #151: Where do the functions biggestSize() or back() or firstAvailable() come from?

siralam, 2017-12-15:
I am so surprised I found no issues asking this question... Everyone seems to be calling biggestSize() happily... Am I missing anything?!

I have gone through the README again and again and I think have completely no idea why in the example you can just use biggestSize() and back(), etc, when configuring Fotoapparat...

siralam, 2017-12-15:
I found it, those are all static functions...

In fact, Android Studio cannot auto complete static functions without Stating its class first. So unless I type the full name of that function and let Android Studio asks me if I want to import that static function or not, I (as I user) will not know there is such a function.

closed 2017-12-15 by siralam

dmitry-zaitsev, 2017-12-15:
We find that that kind of syntax is greatly improving readability of the configuration (assuming that you know where functions are coming from) at the cost of discoverability. 

But thanks for the feedback, I will work on a Wiki for our library to make things more clear.