RedApparat/Fotoapparat #152: Suggestion: stop() should not throw RuntimeException if camera is not started

siralam, 2017-12-15:
This is because when we ask for runtime permission (**before** calling start()), onPause() will be triggered (to show the permission dialog).

Therefore, of course, in such case, camera has not been started. That means you forces your users to try catch`stop()` inside `onPause()` I think this is not a good design.

If your code checked that camera has not been started when calling stop(), why not simply return?

Diolor, 2017-12-15:
You should not call start if you don't have permissions

closed 2017-12-15 by Diolor

Diolor, 2017-12-15:
You should not call stop if you don't have permissions either

dmitry-zaitsev, 2017-12-15:
@siralam we are actually calling camera methods inside `onStart` and `onStop` and do an extra check for the permissions.

As for the exception - I think we might actually remove it from `stop()` and make it no-op in case if camera was not started. @Diolor what do you think?

Diolor, 2017-12-15:
TL:DR; You should not call stop() either if you have no permission. [See sample.](https://github.com/Fotoapparat/Fotoapparat/blob/master/sample/src/main/java/io/fotoapparat/sample/MainActivity.java#L249)

In general you should not interact with FA & camera if you don't have permissions.
Regarding `stop` being partially no-op: will make the code less cryptic and less easy to notice bugs.