RedApparat/Fotoapparat #153: Suggestion: Add callback when focus is locked

siralam, 2017-12-15:
Let's say I want to play a shutter animation when the focus is locked / just before taking a picture.

I don't want to play it together with pressing the capture button because the camera needs time to lock focus.

Hope this feature will be added in the future!

Raenar4k, 2017-12-15:
Hey! While there are no general callback when camera is focused, you can get callback when you are triggering focus manually:
```Kotlin
        fotoapparat.focus().whenAvailable { result: FocusResult? -> 
            Log.d("callback test", "focus now in $result state")
        }
```
But that does not help when taking photos, so i agree, callback could be very useful.
(As a workaround you could manually focus, then take photo, while disabling auto focus which triggers by photo - but that too is not possible yet, see #149 )

Diolor, 2017-12-18:
Plan is to remove the focus routine just before taking a photo (#149). Then you will be able to use the code @Raenar4k perfectly suggested.

I assume this issue solved. Will continue with #149 about takePicture/focus relation

closed 2017-12-18 by Diolor