RedApparat/Fotoapparat #156: Error "Failed to open camera with lens position: [NAME] and id: [ID]"

koenschauwaert, 2017-12-16:
When I just start the app, nothing goes wrong. I can switch between front and rear (I used the Sample code), I can make a picture, go to a new activity and view this picture, but when pressing the back button... 
```java
@Override
public void onBackPressed(){
    super.onBackPressed();
}
```
...to return to the activity where my cameraview is shown, I get the following error:

```java
io.fotoapparat.b.b: Failed to open camera with lens position: BACK and id: 0
```
And
```java
io.fotoapparat.b.b: Failed to open camera with lens position: FRONT and id: 1
```

# Context
Devices:
- Moto X 2013 // API 22 // LineageOS
- Xiaomi Mi5S // API 25 // LineageOS
- Huawei P10 // API 25 // Original EMUI ROM

I use the newest library:
```java
implementation 'io.fotoapparat.fotoapparat:library:1.5.0'
```

Diolor, 2017-12-16:
Can you please paste or upload the code which is not working?

closed 2017-12-29 by Diolor

devangichhatbar, 2018-10-21:
The same issue facing. 
[image]

When I click on the bottom tab and then click on camera there is the blank screen.