RedApparat/Fotoapparat #162: java.lang.NullPointerException Exception on Some Devices

itt80, 2017-12-26:
It's working perfectly. But I get these exceptions on some devices
`Fotoapparat
                .with(this)
                //.cameraProvider(CameraProviders.v2(this)) // change this to v2 to test Camera2 API
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .photoSize(wideRatio(smallestSize()))
                .lensPosition(lensPosition(position))
                .focusMode(firstAvailable(
                        continuousFocus(),
                        autoFocus(),
                        fixed()
                ))
                .previewFpsRange(rangeWithHighestFps())
                .sensorSensitivity(highestSensorSensitivity())
                .frameProcessor(new SampleFrameProcessor())
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new CameraErrorCallback() {
                    @Override
                    public void onError(CameraException e) {
                        Toast.makeText(FotoApparatActivity.this, e.toString(), Toast.LENGTH_LONG).show();
                    }
                })
                .build();`

It's occured this exception messages;

` java.lang.NullPointerException
Attempt to invoke virtual method 'float io.fotoapparat.parameter.Size.getAspectRatio()' on a null object reference

io.fotoapparat.parameter.provider.InitialParametersProvider.previewWithSameAspectRatio (InitialParametersProvider.java:70)
io.fotoapparat.parameter.provider.InitialParametersProvider.validPreviewSizeSelector (InitialParametersProvider.java:61)
io.fotoapparat.parameter.provider.InitialParametersProvider.previewSizeParameters (InitialParametersProvider.java:114)
io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters (InitialParametersProvider.java:84)
io.fotoapparat.routine.StartCameraRoutine.tryToStartCamera (StartCameraRoutine.java:60)
io.fotoapparat.routine.StartCameraRoutine.run (StartCameraRoutine.java:47)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1112)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:587)
java.lang.Thread.run (Thread.java:818)
`

**Library:** 1.5.0

**Devices and Android Api:**
- Samsung Galaxy J3 & 5.1.1
- Samsung Galaxy S3 Mini & 4.1.2
- Samsung Galaxy Core Prime & 5.1.1
- Samsung Galaxy Tab E 9.6 & 4.4.4
- Samsung Galaxy S2 & 4.1.2

itt80, 2017-12-27:
Why are the device models working stable? Are they unsupported?

Diolor, 2017-12-29:
You only select a wide ratio ( `wideRatio(smallestSize())` ) so I assume these devices cannot generate a aspect ratio. Therefore the exception. use `firstAvailable()` selector or catch the error on these devices

closed 2017-12-29 by Diolor