RedApparat/Fotoapparat #163: java.lang.IllegalArgumentException: FocusMode mode manual is not supported. Samsung Grand Prime

sefaaycicek, 2017-12-30:
Hi, I tried different methods with different devices. Camera starts successfully on Samsung Note 5 but it always crash on Samsung Grand Prime(Android 6.0).

**This is create method.**  I set focus mode correctly. 
private Fotoapparat createFotoapparat() {
        return Fotoapparat
                .with(this)
                .into(cameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .lensPosition(back())
                .focusMode(firstAvailable(continuousFocusPicture(),autoFocus(),fixed()))
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new Function1<CameraException, Unit>() {
                    @Override
                    public Unit invoke(CameraException e) {
                        Toast.makeText(TestActivity.this, e.toString(), Toast.LENGTH_LONG).show();
                        return null;
                    }
                })
                .build();
    }

Crash log says that Focus mode manuel is not supported.
- This crash is for Samsung Grand Prime. This code work on Samsung Note 5. 
- We already used 1.5.0 version on same device with same parameters. And it work with no problem. We want to change sdk because of 1.5.0 focus stability problem. 

**Also we tried to set configuration manually as below code.**
        fotoapparat = createFotoapparat();
        fotoapparat.updateConfiguration(cameraConfiguration);

**Configuration instance is here.**
 private CameraConfiguration cameraConfiguration = CameraConfiguration
            .builder()
            .photoResolution(standardRatio(
                    highestResolution()
            ))
            .focusMode(firstAvailable(
                    continuousFocusPicture(),
                    autoFocus(),
                    fixed()

            ))
            .flash(firstAvailable(
                    autoRedEye(),
                    autoFlash(),
                    torch(),
                    off()
            ))
            .previewFpsRange(highestFps())
            .sensorSensitivity(highestSensorSensitivity())
            .build();

Diolor, 2017-12-30:
Hey @sefaaycicek, thanks for opening a separate issue again! Could you also please paste the exact stacktrace? It would help us a lot.

My assumption so far is that `Samsung Grand Prime` violates the possible values of the Android camera Api and also returns `manual` which is not in the [expected set of values](https://developer.android.com/reference/android/hardware/camera2/CaptureRequest.html#CONTROL_AF_MODE). Instead of crashing the app we can simply find a way to ignore this value, but let's check first the stacktrace

Diolor, 2017-12-30:
I made a blind fix. Try `2.0.0-beta3`, should work I think

sefaaycicek, 2017-12-30:
App crashed for another reason. 
Caused by: java.lang.ClassNotFoundException: Didn't find class "kotlinx.coroutines.experimental.CompletableDeferredKt"

```
12-30 18:06:25.333 32619-32619/com.bap.android E/AndroidRuntime: FATAL EXCEPTION: main
                                                                 Process: com.bap.android, PID: 32619
                                                                 java.lang.NoClassDefFoundError: Failed resolution of: Lkotlinx/coroutines/experimental/CompletableDeferredKt;
                                                                     at io.fotoapparat.hardware.Device.<init>(Device.kt:40)
                                                                     at io.fotoapparat.hardware.Device.<init>(Device.kt:27)
                                                                     at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:59)
                                                                     at io.fotoapparat.FotoapparatBuilder.buildInternal(FotoapparatBuilder.kt:174)
                                                                     at io.fotoapparat.FotoapparatBuilder.build(FotoapparatBuilder.kt:163)
                                                                     at com.bap.android.Activity.TestActivity.createFotoapparat(TestActivity.java:116)
                                                                     at com.bap.android.Activity.TestActivity.onCreate(TestActivity.java:94)
                                                                     at android.app.Activity.performCreate(Activity.java:6904)
                                                                     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1136)
                                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3266)
                                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3415)
                                                                     at android.app.ActivityThread.access$1100(ActivityThread.java:229)
                                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1821)
                                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                     at android.os.Looper.loop(Looper.java:148)
                                                                     at android.app.ActivityThread.main(ActivityThread.java:7406)
                                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
                                                                  Caused by: java.lang.ClassNotFoundException: Didn't find class "kotlinx.coroutines.experimental.CompletableDeferredKt" on path: DexPathList[[zip file "/data/app/com.bap.android-2/base.apk"],nativeLibraryDirectories=[/data/app/com.bap.android-2/lib/arm, /data/app/com.bap.android-2/base.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]
                                                                     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
                                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
                                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
                                                                     at io.fotoapparat.hardware.Device.<init>(Device.kt:40) 
                                                                     at io.fotoapparat.hardware.Device.<init>(Device.kt:27) 
                                                                     at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:59) 
                                                                     at io.fotoapparat.FotoapparatBuilder.buildInternal(FotoapparatBuilder.kt:174) 
                                                                     at io.fotoapparat.FotoapparatBuilder.build(FotoapparatBuilder.kt:163) 
                                                                     at com.bap.android.Activity.TestActivity.createFotoapparat(TestActivity.java:116) 
                                                                     at com.bap.android.Activity.TestActivity.onCreate(TestActivity.java:94) 
                                                                     at android.app.Activity.performCreate(Activity.java:6904) 
                                                                     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1136) 
                                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3266) 
                                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3415) 
[... cut ...]
                                                                     at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
                                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
                                                                     		... 20 more
                                                                  Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
```

Diolor, 2017-12-31:
I will assume this fixed with `2.0.0-beta5`. Feel free to comment if you have further problems

closed 2017-12-31 by Diolor