RedApparat/Fotoapparat #164: App is crashing while rotate screen continuously on Samsung galaxy on 5

GrishmaPromact, 2018-01-01:
Today, I just take latest from fotoappart and app is still crashing while rotating screen continuously.
PFA, Crash log screenshot.
[image]

Please do needful.

Diolor, 2018-01-01:
I cannot reproduce neither on Nexus 6 nor on a Pixel 2

GrishmaPromact, 2018-01-01:
But it is crashing in Samsung devices. Please check it in Samsung mobile.

GrishmaPromact, 2018-01-02:
@Diolor Same crash accrued in moto M device.

closed 2018-01-03 by Diolor

GrishmaPromact, 2018-01-03:
@Diolor It is not resolved.. Why are you close my issue?

GrishmaPromact, 2018-01-03:
@Diolor Thanks. This crashed is resolved. Another crash came while doing same. 
PFA, of crashed log.

[image]

commit 2513e428 (2018-01-03): Notify exception from the callback instead of crashing the app. Fixes #164, #165