RedApparat/Fotoapparat #166: Frame processor not working with v2.0.0?

jpribble, 2018-01-02:
#### What are you trying to achieve or the steps to reproduce?
I'm trying to use the frameProcessor() feature to analyze preview frames from the camera.

#### How did you initialize FA?
```java
Fotoapparat
        .with(this)
        .into(cameraView!!)
        .previewScaleType(ScaleType.CenterCrop)
        .photoResolution([custom_selector])
        .previewResolution([custom_selector])
        .lensPosition(single(LensPosition.Back))
        .focusMode(firstAvailable(
            continuousFocusPicture(),
            autoFocus(),
            fixed()))
        .flash(off())
        .frameProcessor {
          processFrame(it)
        }
        .previewFpsRange(highestFps())
        .cameraErrorCallback { e ->
          [error_handling_code]
        }
        .logger(logcat())
        .build()
```

#### What was the result you received?
The frame processor's invoke(frame) callback is never called. Is this feature still a work in progress for version 2.0.0 or should this be working? I noticed that the sample app's Kotlin activity doesn't use a frame processor but the Java activity does. I tried modifying the sample app to use the Java activity but its SampleFrameProcessor didn't receive any callbacks either.

#### What did you expect?
I expected the frame processor to receive callbacks with preview frames from the camera.

#### Context:

* *FA version*: 2.0.0
* *Devices/APIs affected*: Tested on Samsung Galaxy S8 running Android 7.0.

Diolor, 2018-01-02:
It should be working. I will take a look. Thanks for reporting!

Diolor, 2018-01-03:
FotoapparatBuilder had not linked a field. Fixed in 2.0.1 (already out)

closed 2018-01-03 by Diolor