RedApparat/Fotoapparat #167: Camera focus loses when picking multiple image

sefaaycicek, 2018-01-03:
Here is FA instance configuration

```
 return Fotoapparat
                .with(this)
                .into(cameraView)
                .previewScaleType(ScaleType.CenterInside)
                .lensPosition(back())
                .focusMode(firstAvailable(autoFocus(),fixed()))
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new Function1<CameraException, Unit>() {
                    @Override
                    public Unit invoke(CameraException e) {
                        Toast.makeText(BapCameraActivity2.this, e.toString(), Toast.LENGTH_LONG).show();
                        return null;
                    }
                })
                .build();
```

Some of them were blurred when shooting multiple photos.
We try different ways. 

`PhotoResult photoResult = fotoapparat.autoFocus().takePicture();`

```
fotoapparat.focus().whenAvailable(new Function1<FocusResult, Unit>() {
            @Override
            public Unit invoke(FocusResult focusResult) {
                Toast.makeText(BapCameraActivity2.this, focusResult.toString(), Toast.LENGTH_LONG).show();  //. **This log always focused**
                PhotoResult photoResult = fotoapparat.takePicture();
}
}
```

Diolor, 2018-01-04:
Both capturing ways are exactly the same. Maybe the camera was focused but on a different focal length for some reason?
All these operations are sequential when capturing multiple photos (e.g invoking those operations twice) : Focus, Capture, Focus, Capture, ...

sefaaycicek, 2018-01-04:
Yes. when user click capture button, we call focus&capture. 

This issue can be related with focus point? Can you add a method like FocusToCenter, could it be the solution?

Diolor, 2018-01-07:
If the lower-level drive tells us that it's focused we can't really do much here 🤷‍♂️ . We will add #171 soon with manual focus, if that helps.

Diolor, 2018-01-13:
So again if the device says it's focused we can't do much

closed 2018-01-13 by Diolor