RedApparat/Fotoapparat #170: getCurrentParameters()/getCapabilities().whenAvailable{} returns null if called too early

jpribble, 2018-01-05:
#### What are you trying to achieve or the steps to reproduce?
I'm trying to call Fotoapparat.getCurrentParameters().whenAvailable { ... } and Fotoapparat.getCapabilities().whenAvailable { ... } in my Activity's onStart().

```java
override fun onStart() {
  super.onStart()
  camera.start()
  camera.getCurrentParameters().whenAvailable { … }
  camera.getCapabilities().whenAvailable { … }
}
```

#### How did you initialize FA?
```java
override fun onCreate() {
  super.onCreate()
  …
  camera = Fotoapparat
    .with(this)
    .into(cameraView!!)
    .previewScaleType(ScaleType.CenterCrop)
    .photoResolution([custom_selector])
    .previewResolution([custom_selector])
    .lensPosition(single(LensPosition.Back))
    .focusMode(firstAvailable(
        continuousFocusPicture(),
        autoFocus(),
        fixed()))
    .flash(off())
    .frameProcessor { … }
    .previewFpsRange(highestFps())
    .cameraErrorCallback { … }
    .logger(logcat())
    .build()
}
```

#### What was the result you received?
Both calls return NULL and the log shows "Couldn't deliver pending result":
> 01-05 15:48:56.884 9089-9089/[package_name] D/Fotoapparat: Fotoapparat: getCapabilities
> 01-05 15:48:56.887 9089-13521/[package_name] D/Fotoapparat: CameraDevice: getParameters
> 01-05 15:48:56.888 9089-13094/[package_name] D/Fotoapparat: Couldn't deliver pending result.

#### What did you expect?
I expect the calls to return valid results and delay their async callbacks if the camera is still initializing.

#### Context:

* *FA version*: 2.0.2
* *Devices/APIs affected*: Samsung Galaxy S8 running Android 7.0.
* *Other Notes*: This used to work in version 1.5.0.

Diolor, 2018-01-07:
(same will happen for current parameters)

Diolor, 2018-01-16:
Explicit, improved logs in #181

jpribble, 2018-01-18:
I'm still seeing this issue when testing the latest master code. Here are the logs:
> 01-18 10:52:17.208 11349-11349/ D/Fotoapparat: Fotoapparat: <init>
> 01-18 10:52:17.228 11349-11349/ D/Fotoapparat: Fotoapparat: start
> 01-18 10:52:17.228 11349-11349/ D/Fotoapparat: Fotoapparat: getCurrentParameters
> 01-18 10:52:17.228 11349-12587/ D/Fotoapparat: Device: selectCamera
> 01-18 10:52:17.228 11349-12587/ D/Fotoapparat: CameraDevice: open
> 01-18 10:52:17.228 11349-11349/ D/Fotoapparat: Fotoapparat: getCapabilities
> 01-18 10:52:17.228 11349-12975/ D/Fotoapparat: CameraDevice: getParameters$suspendImpl
> 01-18 10:52:17.368 11349-12587/ D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl
> 01-18 10:52:17.368 11349-12587/ D/Fotoapparat: CameraDevice: updateParameters$suspendImpl
> 01-18 10:52:17.368 11349-12975/ D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl
> **01-18 10:52:17.368 11349-12590/ D/Fotoapparat: Couldn't deliver pending result: Operation failed internally.**
> 01-18 10:52:17.368 11349-12587/ D/Fotoapparat: New camera parameters are: CameraParameters ...

dmitry-zaitsev, 2018-01-20:
I believe that pull request will fix it #188

PR #172 (2018-01-08): Wait capabilities + parameters to be created before obtaining them.
Fixes #170
closed 2018-01-20 by dmitry-zaitsev