RedApparat/Fotoapparat #173: Minimum API version changed from 14 to 16

bernatpl, 2018-01-10:
Hello,

I found out that the minimum API version changed from 14 to 16. (https://github.com/Fotoapparat/Fotoapparat/commit/2ac387d6ff77caca6b7f5dd9799e17cd5c3c7f57)
I must support at least API 15, so it blocks me to use the latest features and bug fixes in newer versions.
I didn't see this change documented anywhere. It would be good to document it at least on the release notes.
I tried to build the library using the minimum API as 14 And the sample application works perfectly.
I'm wondering if it was really necessary to change from minimum API 14 to 16.

Thanks!

Diolor, 2018-01-10:
Hey. That was an accidental change actually. We can revert to 14. If you happen make a PR we can quickly make a 2.1.2 patch version

bernatpl, 2018-01-10:
Pull request #174 passed all the checks.

closed 2018-01-10 by Diolor

bernatpl, 2018-01-10:
Thank you very much for fixing it so fast! You're doing a great job here! :)

Diolor, 2018-01-10:
❤️  
Thanks for the PR!