RedApparat/Fotoapparat #177: Samsung S4 Mini - front camera not working?

shahzadyounas, 2018-01-11:
I have tried the latest build 2.1.2...and it no longer works on below:

Samsung BUILD.PRODUCT: serranoltexx (Samsung S4 Mini CyangenMod) Android 4.4.2

I don't get any camera preview. Works fine in emulator and also on HTC M10 no issues.

Any ideas?

`01-11 17:58:45.452 1043-1043/? D/KeyguardHostView: enableCameraWidget : true, cameraWidgetType : 0
01-11 17:58:45.832 1043-1043/? D/SecCameraShortcut: setCallback(): null
01-11 17:58:45.842 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:58:45.842 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:58:45.842 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:58:45.892 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:59:16.675 764-823/? W/PackageManager: Unknown permission com.sec.android.app.twdvfs.DVFS_BOOSTER_PERMISSION in package com.sec.factory.camera
01-11 17:59:16.715 764-823/? W/PackageManager: Unknown permission com.samsung.difactorycamera.permission.ACCESS in package com.sec.android.app.parser
01-11 17:59:16.725 764-823/? W/PackageManager: Unknown permission com.sec.android.app.camera.permission.SHOOTING_MODE in package com.arcsoft.picturesbest.app
01-11 17:59:16.745 764-823/? W/PackageManager: Unknown permission com.samsung.difactorycamera.permission.ACCESS in package com.sec.android.app.hwmoduletest
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission android.permission.ACCESS_DEV_STORAGE in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.sec.android.app.twdvfs.DVFS_BOOSTER_PERMISSION in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.sec.chaton.TOKEN_74e548bcd7dc013c8f8b585ba11ad16b271ceebb0acbe6c001c0aecdae65f37b366963990139710b4b5c3238053d25b7940d90c9f30e04a119c6c3dd94c276d32ec19cc4a4906ccbae88b80158ce08440ba5b1d7e14b864a076e31fa3e478f51ff186b2e995cc4d9503987cc3b80efdc331d7626ae145ce1810604514abd5fd9 in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.samsung.android.provider.filterprovider.permission.READ_FILTER in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.samsung.android.provider.filterprovider.permission.WRITE_FILTER in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.sec.android.daemonapp.cmaweather.CMAWEATHER_DAEMON_ACCESS_PROVIDER in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.sec.android.daemonapp.ap.kweather.KWEATHER_DAEMON_ACCESS_PROVIDER in package com.sec.android.app.camera
01-11 17:59:16.755 764-823/? W/PackageManager: Unknown permission com.sec.android.daemonapp.ap.weathernewsjp.WEATHERNEWS_DAEMON_ACCESS_PROVIDER in package com.sec.android.app.camera
01-11 17:59:38.799 1043-1043/? D/KeyguardHostView: enableCameraWidget : true, cameraWidgetType : 0
01-11 17:59:39.550 1043-1043/? D/SecCameraShortcut: setCallback(): null
01-11 17:59:39.560 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:59:39.560 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:59:39.560 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 17:59:39.600 1043-1043/? D/SecCameraShortcut: setCallback(): not null
01-11 18:00:23.446 198-417/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:23.496 198-198/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:23.837 198-1000/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:23.867 198-417/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:23.897 764-1756/? I/EDMNativeHelperService: isCameraEnabled
01-11 18:00:23.897 198-198/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:23.897 198-198/? E/QualcommCamera: Qint android::get_camera_info(int, camera_info*): E
01-11 18:00:24.057 198-198/? D/SecCameraCoreManager: SecCameraCoreManager created: pid=198, cameraId=1
01-11 18:00:24.057 198-198/? E/SecCameraCoreManager: !@ cam: CaMeRa(1)
01-11 18:00:24.067 198-198/? D/SecCameraCoreManager: initialize
01-11 18:00:24.067 198-198/? I/SecCameraCoreManager: Opening camera 1
01-11 18:00:24.067 198-198/? E/QualcommCamera: Qint android::camera_device_open(const hw_module_t*, const char*, hw_device_t**): HAL_currentCameraId : 1 E#######
01-11 18:00:24.067 198-198/? I/QCameraHWI: QCameraHardwareInterface: E
01-11 18:00:24.067 198-198/? I/QCameraHWI: Init mPreviewState = 0
01-11 18:00:24.067 198-198/? E/QCameraHWI: startCamera: camera_open: mCameraId = 1
01-11 18:00:24.067 198-198/? E/mm-camera-intf: camera_open: E camera_idx = 1
01-11 18:00:24.067 198-198/? E/mm-camera-intf: mm_camera_open:  begin cam_idx=1 my_obj->my_hdl=2817
01-11 18:00:24.087 420-20789/? E/mm-camera: Samsung cam_mctl_thread: set camera_id =1
01-11 18:00:24.087 420-20789/? E/mm-camera: mctl_store_curr_target Current target is 1 id = 154
01-11 18:00:24.087 420-420/? E/mm-camera: camera_plugin_client_open: plugin client open exit, rc = 0
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_open:  ctrl_fd = 48, errno == 22
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_open:  opened, break out while loop
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_socket_create:Success socket_fd=49 /data/cam_socket1
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_open:  ds_fd = 49, errno = 22
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_open:  opened, break out while loop
01-11 18:00:24.087 198-198/? E/mm-camera-intf: mm_camera_poll_thread_launch: poll_type = 0, read fd = 51, write fd = 53 timeout = 7000, obj_address = 0xb8d7e6c8, poll_cb = 0xb8d7f8f0
01-11 18:00:24.087 198-198/? E/QCameraHWI: Cam open returned 0xb8d7faf0
01-11 18:00:24.097 420-20789/? E/mm-camera: sensor_load_chromatix: libchromatix_s5k6a3yx_preview.so: 32
01-11 18:00:24.107 420-20789/? E/mm-camera: vendor/qcom/proprietary/mm-camera/server/chromatix/0208/libchromatix/chromatix_s5k6a3yx/preview/chromatix_s5k6a3yx_preview.c:load_chromatix:57
01-11 18:00:24.107 420-20789/? E/mm-camera: load_chromatix:61: chromatix_version=520
01-11 18:00:24.107 420-20789/? E/mm-camera: #QCT 1 # chromatix_s5k6a3yx_preview.dat : open -1 
01-11 18:00:24.107 420-20789/? E/mm-camera: load_chromatix:93: chromatix_version=520
01-11 18:00:24.107 420-20789/? E/mm-camera: camera_plugin_client_private_ioctl_to_plugin: received private ioctl
01-11 18:00:24.107 420-20789/? E/mm-camera: camera_plugin_grant_pix_interface Support simultaneous camera 0 Sensor type 0 BAYER 0
01-11 18:00:24.107 420-20789/? E/mm-camera: camera_plugin_grant_pix_interface Support simultaneous camera 0 Sensor type 0
01-11 18:00:24.107 420-20789/? E/mm-camera: SADA  camera_plugin_grant_pix_interface 0
01-11 18:00:24.107 420-20789/? E/mm-camera: axi_client_init axi client 0
                                             
[... 504 lines cut ...]
01-11 18:02:01.171 198-20994/? E/QCameraHWI: Stop recording issued. Return from process Event
01-11 18:02:08.178 198-20998/? E/mm-camera-intf: mm_camera_poll_fn: MM_CAMERA_CH_EVT_STREAMING_ERR
01-11 18:02:08.178 198-20998/? E/mm-camera-intf: mm_camera_poll_fn: poll type = 1, num_fd = 1 poll_cb = 0xb8d5a600, my_obj addr = 0xb8d5a4d0, timeout = 7000
01-11 18:02:08.178 198-20994/? E/QCameraHWI: processEvent: type :0 E
01-11 18:02:08.178 198-20994/? E/QCameraHWI: Stop recording issued. Return from process Event
01-11 18:02:15.186 198-20998/? E/mm-camera-intf: mm_camera_poll_fn: MM_CAMERA_CH_EVT_STREAMING_ERR
01-11 18:02:15.186 198-20998/? E/mm-camera-intf: mm_camera_poll_fn: poll type = 1, num_fd = 1 poll_cb = 0xb8d5a600, my_obj addr = 0xb8d5a4d0, timeout = 7000
01-11 18:02:15.186 198-20994/? E/QCameraHWI: processEvent: type :0 E
01-11 18:02:15.186 198-20994/? E/QCameraHWI: Stop recording issued. Return from process Event
`

I load the camera as so:

`  fotoapparat = Fotoapparat
                    .with(this)
                    .into(cameraPreview)
                    .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                    .lensPosition(front())       // we want back camera

                    .focusMode(firstAvailable(  // (optional) use the first focus mode which is supported by device
                            continuousFocusPicture(),
                            autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                            fixed()             // if even auto focus is not available - fixed focus mode will be used
                    ))
                    .build();`

Diolor, 2018-01-11:
You say `it no longer works`. From which version did you update?

Can you please attach a logcat logger? The above logs are not really useful, sorry.

You can do so adding this in the Builder:
`.logger(logcat())`

shahzadyounas, 2018-01-11:
Confirmed working in v1.4.1

Will get the logs to you tomorrow if that helps!

tx

shahzadyounas, 2018-01-12:
```
01-12 10:30:58.047 764-799/? D/Sensors: LightSensor readEvents x = 200.000000, raw = 200
01-12 10:30:58.148 764-812/? W/ProcessCpuTracker: Skipping unknown process pid 8849
01-12 10:30:58.248 764-799/? D/Sensors: LightSensor readEvents x = 200.000000, raw = 200
01-12 10:30:58.378 198-405/? D/AudioStreamOutALSA: standby
01-12 10:30:58.378 198-405/? D/ALSAModule: s_standby: handle 0xb8d172c0 h 0x0
01-12 10:30:58.378 198-405/? E/ALSAModule: s_standby handle h 0xb8d3c7d8
01-12 10:30:58.448 764-799/? D/Sensors: LightSensor readEvents x = 207.000000, raw = 207
01-12 10:30:58.478 1043-1043/? D/STATUSBAR-NetworkController: refreshSignalCluster - setNWBoosterIndicators(false)
01-12 10:30:58.478 1043-1043/? D/STATUSBAR-NetworkController: refreshSignalCluster: data=-1 bt=false
01-12 10:30:58.488 1043-1043/? D/STATUSBAR-IconMerger: checkOverflow(288), More:false, Req:false Child:7
01-12 10:30:58.548 1043-1043/? D/STATUSBAR-NetworkController: onSignalStrengthsChanged signalStrength=SignalStrength: 22 -1 -1 -1 -1 -1 -1 99 2147483647 2147483647 2147483647 -1 2147483647 gsm|lte 0x4 level=4
01-12 10:30:58.558 198-405/? D/ALSAModule: checkRunningHandle return false
01-12 10:30:58.558 198-405/? D/alsa_ucm: snd_use_case_set(): uc_mgr 0xb8cf2f20 identifier _verb value Inactive
01-12 10:30:58.558 198-405/? D/alsa_ucm: Set mixer controls for HiFi enable 0
01-12 10:30:58.558 198-405/? D/alsa_ucm: Setting mixer control: PRI_RX Audio Mixer MultiMedia1, value: 0
01-12 10:30:58.558 198-405/? E/ALSAModule: Number of modifiers 0
01-12 10:30:58.558 198-405/? E/ALSAModule: usecase_type is 0
01-12 10:30:58.558 198-405/? D/alsa_ucm: snd_use_case_set(): uc_mgr 0xb8cf2f20 identifier _disdev value Speaker
01-12 10:30:58.558 198-405/? D/alsa_ucm: Set mixer controls for Speaker enable 0
01-12 10:30:58.558 198-405/? D/alsa_ucm: Setting mixer control: RX4 MIX1 INP1, value: ZERO
01-12 10:30:58.558 198-405/? D/alsa_ucm: Setting mixer control: SLIM RX1 MUX, value: ZERO
01-12 10:30:58.558 198-405/? D/alsa_ucm: Setting mixer control: SPK DAC Switch, value: 0
01-12 10:30:58.558 198-405/? D/alsa_ucm: Setting mixer control: RX4 Digital Volume, value: 0
01-12 10:30:58.568 198-405/? D/alsa_ucm: Setting mixer control: SPK DRV Volume, value: 0
01-12 10:30:58.568 764-829/? I/ServiceKeeper: In getpackagename pid = 764 uid = 1000 package name = android
01-12 10:30:58.648 764-799/? D/Sensors: LightSensor readEvents x = 197.000000, raw = 197
01-12 10:30:58.848 764-799/? D/Sensors: LightSensor readEvents x = 194.000000, raw = 194
01-12 10:30:59.049 764-799/? D/Sensors: LightSensor readEvents x = 184.000000, raw = 184
01-12 10:30:59.229 764-953/? I/InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.758 ] when=62987440840000
01-12 10:30:59.229 764-952/? I/InputDispatcher: Delivering touch to: action: 0x4, toolType: 1
01-12 10:30:59.229 764-952/? I/InputDispatcher: Delivering touch to: action: 0x0, toolType: 1
01-12 10:30:59.229 764-829/? I/ServiceKeeper: In getpackagename pid = 764 uid = 1000 package name = android
01-12 10:30:59.249 764-799/? D/Sensors: LightSensor readEvents x = 184.000000, raw = 184
01-12 10:30:59.249 764-953/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=62987469010000
01-12 10:30:59.249 764-952/? I/InputDispatcher: Delivering touch to: action: 0x1, toolType: 1
01-12 10:30:59.259 764-1236/? E/DatabaseUtils: Writing exception to parcel
                                               java.lang.SecurityException: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
                                                   at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15174)
                                                   at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2492)
                                                   at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
                                                   at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
                                                   at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
                                                   at android.os.Binder.execTransact(Binder.java:404)
                                                   at dalvik.system.NativeStart.run(Native Method)
01-12 10:30:59.259 764-1236/? W/ActivityManager: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
01-12 10:30:59.259 764-1384/? D/CustomFrequencyManagerService: acquireDVFSLockLocked : type : DVFS_MIN_LIMIT  frequency : 1242000  uid : 1000  pid : 764  pkgName : ACTIVITY_RESUME_BOOSTER@5
01-12 10:30:59.259 764-1384/? W/ActivityManager: mDVFSHelper.acquire()
01-12 10:30:59.279 764-810/? D/PointerIcon: setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
01-12 10:30:59.279 764-810/? D/PointerIcon: setMouseCustomIcon IconType is same.101
01-12 10:30:59.279 764-1064/? D/Sensors: setDelay: acc_delay=200000000, mag_delay=10000000
01-12 10:30:59.279 764-1064/? D/Sensors: setDelay: handle=0, delay=200000000
01-12 10:30:59.279 8068-8068/com.muzmatch.muzmatchapp D/SensorManager: unregisterListener ::   
[... cut ...]
01-12 10:31:19.450 764-799/? D/Sensors: LightSensor readEvents x = 213.000000, raw = 213
01-12 10:31:19.651 764-799/? D/Sensors: LightSensor readEvents x = 213.000000, raw = 213
01-12 10:31:19.681 1043-1043/? D/STATUSBAR-NetworkController: refreshSignalCluster - setNWBoosterIndicators(false)
01-12 10:31:19.681 1043-1043/? D/STATUSBAR-NetworkController: refreshSignalCluster: data=-1 bt=false
01-12 10:31:19.711 1043-1043/? D/STATUSBAR-IconMerger: checkOverflow(288), More:false, Req:false Child:7

```

shahzadyounas, 2018-01-12:
Is that helpful?

 ```
 fotoapparat = Fotoapparat
                    .with(this)
                    .into(cameraPreview)
                    .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                    .lensPosition(front())       // we want front camera

                    .focusMode(firstAvailable(  // (optional) use the first focus mode which is supported by device
                            continuousFocusPicture(),
                            autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                            fixed()             // if even auto focus is not available - fixed focus mode will be used
                    ))
                    .logger(logcat())
                    .build();
```

If I run with back() as the lensposition all is fine.

Diolor, 2018-01-12:
Not really, sorry. Please filter only your application's logs. We care about `Fotoapparat` tags in the log. 
In the above logs there are only 5-6, which are unfortunately not enough. All the other log lines are noise.

f0restgump92, 2018-01-14:
This is not lens orientation problem, I faced exactly the same issue with Samsung s8. In my case, problem was focus settings since front and back camera have different focus capabilities. 

If you use 

` .focusMode(firstAvailable(continuousFocusPicture(), autoFocus(),fixed()))`

And if any of the given options are not supported on the front camera, it will fail to boot up the preview.

Diolor, 2018-01-14:
@f0restgump92 Yes. It will throw an exception if the config cannot be set on a target camera. Not sure if that's @shahzadyounas ' issue too. I will try find an S4 too

shahzadyounas, 2018-01-16:
Hi Guys

So I tried what you said...still no luck

Output:

```
01-16 15:22:47.068 9936-9936/com.muzmatch.muzmatchapp D/Fotoapparat: Device: updateLensPositionSelector
01-16 15:22:47.068 9936-9936/com.muzmatch.muzmatchapp D/Fotoapparat: Fotoapparat: <init>
01-16 15:22:47.078 9936-9936/com.muzmatch.muzmatchapp D/Fotoapparat: Fotoapparat: start
01-16 15:22:47.078 9936-10079/com.muzmatch.muzmatchapp D/Fotoapparat: Device: selectCamera
01-16 15:22:47.108 9936-10079/com.muzmatch.muzmatchapp D/Fotoapparat: CameraDevice: open
01-16 15:22:47.388 9936-10079/com.muzmatch.muzmatchapp D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl
```

I took out the focus mode code totally:

```
 fotoapparat = Fotoapparat
                    .with(this)
                    .into(cameraPreview)
                    .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                    .lensPosition(front())       // we want front camera

                    .logger(logcat())
                    .build();
```

Exception errors:

```
01-16 15:23:47.262 187-187/? E/SMD: DCD ON
01-16 15:23:47.953 789-1059/? E/DatabaseUtils: Writing exception to parcel
                                               java.lang.SecurityException: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
                                                   at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15174)
                                                   at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2492)
                                                   at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
                                                   at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
                                                   at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
                                                   at android.os.Binder.execTransact(Binder.java:404)
                                                   at dalvik.system.NativeStart.run(Native Method)
01-16 15:23:47.993 9936-9936/com.muzmatch.muzmatchapp E/dalvikvm: Could not find class 'android.support.v4.app.ActivityCompat$SharedElementCallback23Impl', referenced from method android.support.v4.app.ActivityCompat.setEnterSharedElementCallback
01-16 15:23:47.993 9936-9936/com.muzmatch.muzmatchapp E/dalvikvm: Could not find class 'android.support.v4.app.ActivityCompat$SharedElementCallback21Impl', referenced from method android.support.v4.app.ActivityCompat.setEnterSharedElementCallback
01-16 15:23:47.993 9936-9936/com.muzmatch.muzmatchapp E/dalvikvm: Could not find class 'android.support.v4.app.ActivityCompat$SharedElementCallback23Impl', referenced from method android.support.v4.app.ActivityCompat.setExitSharedElementCallback
01-16 15:23:47.993 9936-9936/com.muzmatch.muzmatchapp E/dalvikvm: Could not find class 'android.support.v4.app.ActivityCompat$SharedElementCallback21Impl', referenced from method android.support.v4.app.ActivityCompat.setExitSharedElementCallback
01-16 15:23:48.043 191-277/? E/qdmemalloc: heap_msk=3000000 flags=1
                                            
01-16 15:23:48.043 191-277/? E/qdmemalloc: heap_msk=40000000 flags=1
                                            
01-16 15:23:48.614 194-566/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-566/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-566/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-566/? E/QualcommCamera: Qvoid android::stop_preview(camera_device*): E
01-16 15:23:48.614 194-581/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qvoid android::disable_msg_type(camera_device*, int32_t): E
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qvoid android::stop_preview(camera_device*): E
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qint android::cancel_picture(camera_device*): E
01-16 15:23:48.614 194-194/? E/mm-camera-intf: mm_channel_fsm_fn_stopped: invalid state (1) for evt (11)
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qvoid android::release(camera_device*): E
01-16 15:23:48.614 194-194/? E/QualcommCamera: Qint android::close_camera_device(hw_device_t*): device =0xb71d22f0 E
01-16 15:23:48.614 194-194/? E/mm-still-omx: mm_jpeg_close: E
01-16 15:23:48.614 194-194/? E/mm-still-omx: mm_jpeg_close: abort ongoing jobs
01-16 15:23:48.614 194-194/? E/mm-still-omx: mm_jpeg_close: abort todo jobs
01-16 15:23:48.614 194-194/? E/mm-still-omx: mm_jpeg_close: abort done jobs in cb threads
01-16 15:23:48.614 194-194/? E/mm-still-omx: mm_jpeg_close: X
01-16 15:23:48.614 194-194/? E/QCameraHWI: ~QCameraHardwareInterface: Jpeg closed, rc = 0, mJpegClientHandle = 300
01-16 15:23:48.614 194-194/? E/QCameraHWI_Preview: release : BEGIN
01-16 15:23:48.614 194-194/? E/QCameraStream: streamOff: Stream:0 is not active
01-16 15:23:48.614 194-194/? E/QCameraHWI_Preview: release: END
[... 338 lines cut ...]
[... cut ...]
01-16 15:25:00.841 1095-10698/? E/CscParser: update(): xml file exist
01-16 15:25:00.871 10694-10694/? E/dalvikvm: >>>>> Normal User
01-16 15:25:00.871 10694-10694/? E/dalvikvm: >>>>> com.sec.factory [ userId:0 | appId:1000 ]
01-16 15:25:01.772 10712-10712/? E/dalvikvm: >>>>> Normal User
01-16 15:25:01.772 10712-10712/? E/dalvikvm: >>>>> com.sec.android.app.music:service [ userId:0 | appId:10131 ]
01-16 15:25:02.132 789-1061/? E/Watchdog: !@Sync 28
01-16 15:25:02.282 187-187/? E/SMD: DCD ON

```

f0restgump92, 2018-01-16:
Aha, you don't have Camera permission 

` java.lang.SecurityException: Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
                                                   at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15174)
                                                   at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2492)
                                                   at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
                                                   at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
                                                   at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
                                                   at android.os.Binder.execTransact(Binder.java:404)
                                                   at dalvik.system.NativeStart.run(Native Method)`

shahzadyounas, 2018-01-16:
It does - this is an android 4.4.2 - where permissions are not required for camera specifically.  Looking at the App Info for the app itself also shows it has permission to take photos/cameras.  The above error is unrelated (it is some weird android issue which appears alot)

f0restgump92, 2018-01-16:
Ah sorry i though it was Android complaining about camera permissions, didnt read the details. Can you create a sample project, and il try to reproduce it later tonight

Diolor, 2018-01-16:
Ok looks like it fails to open the camera.
Can u add `cameraErrorCallback` with a logger?

e.g. in Kotlin:
`cameraErrorCallback = { Log.e(LOGGING_TAG, "Camera error: ", it) }`
or java:
`.cameraErrorCallback({it -> Log.e(LOGGING_TAG, "Camera error: ", it) })`

Diolor, 2018-01-18:
@shahzadyounas I saw you made some comments in #55. I assume you found the issue and I can mark this as closed, right?

shahzadyounas, 2018-01-18:
Got it working -

Added the rest of the focus modes and job done!

```
 .focusMode(firstAvailable(  // (optional) use the first focus mode which is supported by device
                            continuousFocusPicture(),
                            autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                            fixed() ,            // if even auto focus is not available - fixed focus mode will be used
                            infinity(),
                            edof(),
                            macro()
                    ))
```

Tx for all the pointers re logging and for the speedy responses guys.

closed 2018-01-18 by shahzadyounas

Diolor, 2018-01-18:
❤️  
I will also add `infinity()` in the default Camera configuration. I though no device used it solely. I was wrong..