RedApparat/Fotoapparat #182: Why do you bring in both Kotlin and java together and make it a little complex

elementlo, 2018-01-18:
#### What are you trying to achieve or the steps to reproduce?

I am using this pretty good library for camera sdk developing , recently I used the old version but now it changes a lot , for the most inconvenient thing is that now I need make my project support Kotlin. To install the AS Kotlin plugin or so on , I think it is better to provide two editions one is pure java the other is the mix one . I'll tell you what I met when using java below:

#### How did you initialize FA?

```java
private Fotoapparat createFotoapparat() {
		return Fotoapparat
				.with(context)
				//.cameraProvider(CameraProviders.v1()) // change this to v2 to test Camera2 API
				.into(cameraView)
				.previewScaleType(ScaleType.CenterCrop)
				.lensPosition(LensPositionSelectorsKt.back())
				.logger(loggers(
						logcat(),
						fileLogger(context)
				))
				.cameraErrorCallback(new Function1<CameraException, unit>() {
					@Override
					public Unit invoke(CameraException e) {
						Toast.makeText(ActivityJava.this, e.toString(), Toast.LENGTH_LONG).show();
						return null;
					}
				})
				.build();
	}
```

#### What was the result you received?
you can see without kotlin support I used "LensPositionSelectorsKt.back()" here because the default "back" is a kotlin class which I can not reach , moreover in the "cameraErrorCallback" I can not import class "unit" because it is also from a kotlin class , I am helpless and the java configuration in the wiki is outdated for it was using the old version of FA . Hope you could do better : )

#### Context:

* *FA version*:2.1.2
* *Devices/APIs affected*:
* *any other relevant information*:

dmitry-zaitsev, 2018-01-18:
Thanks for bringing this up. Indeed, we should enhance the documentation.

However, as for Kotlin support - you don't actually need to include Kotlin support to your own project. The library is supposed to work this way (since Kotlin compiles to the save JVM bytecode in the end).

elementlo, 2018-01-18:
thanks for your contribution , I'll be waiting for the new document .

closed 2018-01-18 by elementlo

Diolor, 2018-01-19:
Hi @elementlo! `the java configuration in the wiki is outdated for it was using the old version of FA`. I don't think this is accurate. Can you please tell which part is outdated?

I have added what @dmitry-zaitsev said above in the wiki and we will also put explicit interfaces for `cameraErrorCallback` and `CameraErrorCallback` to not confuse you

elementlo, 2018-01-23:
Hi,for example,when initialize FA,the method ".photoSize(biggestSize()) "is not available here , such things like this, and when you remove the kotlin support from the sample's gradle script , something would be told not found.