RedApparat/Fotoapparat #190: Recommended way to install fotoapparat

cqql, 2018-01-20:
This library loks great, but I have looked for it on both mavencentral and jcenter but to no avail. Is [including the repo as a submodule](https://stackoverflow.com/questions/18748436/is-it-possible-to-declare-git-repository-as-dependency-in-android-gradle) the recommended way to install and use it?

Thanks, Marten

Diolor, 2018-01-20:
Hi Marten, the recommended config is displayed in our [README](https://github.com/Fotoapparat/Fotoapparat#set-up). You need to add `jitpack` repository

closed 2018-01-20 by Diolor

cqql, 2018-01-20:
I searched for `install`, googled, read the wiki and still managed to gloss over this section. Thank you!

Diolor, 2018-01-20:
🙂

cqql, 2018-01-20:
I just googled the difference between `compile` and `implementation` in gradle files. Are you aware that [`compile` is deprecated](https://youtu.be/7ll-rkLCtyk?t=29m20s)?

Diolor, 2018-01-20:
Thanks! Indeed we use `compile` last time I tried use `api` (so as the end dev has some necessary classes) the it was not behaving great. Possibly because back then we have the gradle tools `3.1.0-alphaX`. Will give a try again when tools `3.1.x` are stable