RedApparat/Fotoapparat #192: Is it possible to set sensorSensitivity back to auto?

andrewcking, 2018-01-24:
I might be missing something obvious but I don't seem to see a way to set the sensorSensitivity back to 'auto' after it has been set to a specific value. Is this currently possible?

Diolor, 2018-01-24:
You are correct. Unfortunately once camera is set a with a value, this cannot be undone. But I can try again here set `null` and maybe that results to clear the value (I think I had tried and camera api was throwing exception)

andrewcking, 2018-01-24:
As you suspected it looks like `null` results in an exception and an improper value such as `-1` gets ignored. 

Out of curiosity is there a minimal commit or comment on an issue that could demonstrate introducing a particular camera parameter to fotoapparat? I haven't kept up super well since the switch to kotlin. It might be helpful if I get some time to play around with this but I am also thinking that exposure may actually better suit my use case. I am not entirely sure where to start but I imagine the parameters will all follow a similar pattern.

Diolor, 2018-01-25:
hey. I will close then this issue as "unfortunately you cannot reset sensitivity".

This [commit](https://github.com/Fotoapparat/Fotoapparat/commit/1ca2462f0777364f827499366ba8712fa5467ae0) adds anti-banding mode and it's a good example. #39 and #40 are good and fairly easy topics to start with if you like. And yeah, all params follow the same pattern and "code path".

Feel free to play around and we can help you in the PR if something is uncertain.

closed 2018-01-25 by Diolor