RedApparat/Fotoapparat #193: Front facing camera always rotates the result

SAGARSURI, 2018-01-26:
I am using the latest version of your library. I am using only front facing camera. I took the code from your sample app. But when I take the picture from front facing camera, the result is rotated and zoomed. I have to rotate the result bitmap to 270 degree and then flip it to get the proper result. May I know the actual issue and How can it be fixed ? Here is my code:

```
private Fotoapparat createFotoapparat() {
        return Fotoapparat
                .with(this)
                .photoResolution(standardRatio(
                    highestResolution()
                ))
                .into(cameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .lensPosition(front())
                .focusMode(firstAvailable(
                        continuousFocusPicture(),
                        autoFocus(),
                        fixed()
                ))
                .flash(firstAvailable(
                        autoRedEye(),
                        autoFlash(),
                        torch(),
                        off()
                ))
                .previewFpsRange(highestFps())
                .sensorSensitivity(highestSensorSensitivity())
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new CameraErrorListener() {
                    @Override
                    public void onError(CameraException e) {
                        Toast.makeText(CameraActivity.this, e.toString(), Toast.LENGTH_LONG).show();
                    }
                })
                .build();
    }
```

Diolor, 2018-01-26:
Hi, thanks for reporting this. 
Which phone/(s) is affected? Is the sample app affected too?

SAGARSURI, 2018-01-26:
Samsung galaxy s8, Google pixel. Both running on 7.0

Diolor, 2018-01-26:
Is the sample app affected too?

SAGARSURI, 2018-01-26:
In the sample app its flipping the image. But in my case its rotating as well as flipping. I am using the same code from your sample app.

andrewcking, 2018-01-29:
I am also seeing this on my devices. I don't think it is specific to the front facing camera though, rather it is something I see occurring after resuming the app while in landscape mode.

1. Open the sample fotoapparat app 
2. Rotate the device into landscape mode
3. Change to front facing camera
4. Take photo and see incorrectly rotated result

Step # 3 can be replaced with any thing that switches or reinitializes the activity while in landscape mode. For instance, if you rotate to landscape, hit your recents button then return to the fotoapparat sample app and take a photo with the back facing camera the result will be rotated.

I would guess this might be due to an issue with #90 assuming the default orientation is portrait at some point and not updating the orientation to landscape because no orientation update triggers it. 

For what its worth, I actually would prefer it if fotoapparat had the option to let me manage my own rotation. For my use case I don't allow my camera activity into all 4 orientations (I think most camera apps prevent the upsidedown/reverse portrait orientation). So now if a user takes a photo with the phone slightly tilted in an orientation I do not allow, I need to get the rotation and correct it. Previously I could dependably rely on the activity orientation.

Diolor, 2018-01-29:
I think the issue in the video is different than the one in the reproduction steps.
There reason for the first might be as @andrewcking decribed the updated orientation state is lost.
For the second just a wrong calculation.

ptrbrynt, 2018-02-02:
@Diolor Also having this issue on Galaxy Tab S2. Taking a front-facing photo with the device in landscape orientation is fine, but portrait orientation photos are rotated by 90 degrees clockwise.

andrewcking, 2018-02-02:
In my testing I believe the issue lies with this line in `routine/camera/StartRoutine.kt`:

`deviceOrientation = Orientation.Vertical.Portrait,`

If I modify this to `getScreenOrientation` it works but Landscape and ReverseLandscape are switched. If I switch those around in `getScreenOrientation` then the camera preview shows up upsidedown but the photos come in correctly in all the above tests. Not sure what a clean solution is but I thought this might be helpful to someone.

emilany, 2018-02-05:
@Diolor Having the same issue with a BV6000s. How can this be addressed?

Diolor, 2018-02-05:
@emilany The calculations inside `OrientationResolver#computeImageOrientation` need to be fixed

xxxifan, 2018-03-07:
I noticed that when I specify camera resolution parameter, like I want width:1080xheight:1440, but I have to set width=1440, height=1080,  is this one of reason why it change orientation?

andrewcking, 2018-03-27:
For anyone curious, I was able to resolve the issue for my purposes by setting `Device.startOrientationMonitoring` in `StartOrientationRoutine.kt` to just be an empty method. This perhaps isn't a great practice but it prevents fotoapparat from trying to correct orientation based on the way you are holding the device. I then am handling the rotation myself (as was the case pre version 2) and saving the file myself. 

I use the following for rotating the image:

```
  Matrix matrix = new Matrix();
  if (camera.equals("Front")) {
    matrix.postRotate(amountRotated-result.rotationDegrees);
  } else {
    matrix.postRotate(-amountRotated-result.rotationDegrees);
  }
  Bitmap rotated = Bitmap.createBitmap(result.bitmap, 0, 0, result.bitmap.getWidth(), result.bitmap.getHeight(), matrix, true);
```

Where `amountRotated` is the device orientation offset, calculated however you want (using your own orientation listener or basing it on activity orientation). Doing it this way allows me to prevent photos from being taken in certain orientations. 

This seems to be working on front and back cameras and in all orientations on my devices including those with non-standard camera mountings like the Nexus 5x.

dineshvg, 2018-09-14:
@andrewcking How do you obtain the amoutRotated value?

XinyueZ, 2018-09-26:
I rotate bitmap and flip it while processing frames.

```kotlin
   override fun process(frame: Frame) {
        Log.d(TAG, "frame: ${frame.image.size}, ${frame.size}")

        YuvImage(
            frame.image,
            ImageFormat.NV21,
            frame.size.width,
            frame.size.height,
            null
        ).let { yuvImage ->
            ByteArrayOutputStream().use { output ->
                yuvImage.compressToJpeg(
                    Rect(0, 0, frame.size.width, frame.size.height),
                    100,
                    output
                )
                output.toByteArray().apply {
                    BitmapFactory.decodeByteArray(this, 0, size)?.let { bitmap ->
                        process(
                            bitmap.rotate(90f).flip(
                                false, activeCamera == Camera.Front
                            )
                        )
                        bitmap.recycle()
                    }
                }
            }
        }
    }

    private fun Bitmap.flip(horizontal: Boolean, vertical: Boolean): Bitmap {
        val matrix = Matrix()
        matrix.preScale((if (horizontal) -1 else 1).toFloat(), (if (vertical) -1 else 1).toFloat())
        return Bitmap.createBitmap(this, 0, 0, width, height, matrix, true)
    }

    private fun Bitmap.rotate(degrees: Float): Bitmap {
        val matrix = Matrix()
        matrix.postRotate(degrees)
        val scaledBitmap = Bitmap.createScaledBitmap(this, width, height, true)
        return Bitmap.createBitmap(
            scaledBitmap,
            0,
            0,
            scaledBitmap.width,
            scaledBitmap.height,
            matrix,
            true
        )
    }

```

commit 1783058a (2019-03-02): Fixes orientation issues (#354)

* fix: preview orientation in sensorLandscape

* Fix orientation issues.

Fixes #262
Fixes #260
Fixes #253
Fixes #193
closed 2019-03-02 by Diolor