RedApparat/Fotoapparat #196: "Unchecked generic array creation for varargs parameter" in builder for "firstAvailable"

scifinder, 2018-01-28:
Hello! =)
I'm updated Fotoapparat in my application from 1.3.0 to 2.2.0. I rewrote this code fragment
```
        fotoapparat = Fotoapparat
                .with(mContext)
                .into(cameraView)
                .previewSize(
                        firstAvailable(
                                aspectRatio(
                                        16f/9f,
                                        smallestSize()
                                ),
                                aspectRatio(
                                        4f/3f,
                                        smallestSize()
                                )
                        )
                )
                .photoSize(
                        firstAvailable(
                                aspectRatio(
                                        16f/9f,
                                        biggestSize()
                                ),
                                aspectRatio(
                                        4f/3f,
                                        biggestSize()
                                )
                        )
                )
                .flash(
                        isFlash ? on() : off()
                )
                .build();
```
like this
```
        fotoapparat = Fotoapparat
                .with(mContext)
                .into(cameraView)
                .previewResolution(
                        firstAvailable(
                                aspectRatio(16f/9f, lowestResolution()),
                                aspectRatio(4f/3f, lowestResolution())
                        )
                )
                .photoResolution(
                        firstAvailable(
                                aspectRatio(16f/9f, highestResolution()),
                                aspectRatio(4f/3f,highestResolution())
                        )
                )
                .flash(
                        isFlash ? on() : off()
                )
                .build();
```
But I get a warning "Unchecked generic array creation for varargs parameter" for "firstAvailable". This is normal?

Diolor, 2018-01-29:
Yes it is because how Java deals with generics.

Ideas for fixing/suppressing this error: https://medium.com/@BladeCoder/fixing-ugly-java-apis-read-only-generic-varargs-ee2d2e464ac1

scifinder, 2018-01-29:
Ok, I append @SuppressWarnings("unchecked") before my function.

closed 2018-02-01 by Diolor

peter-palmer, 2021-01-10:
Use `@SafeVarargs` before the initialization of the function.

PR #202 (2018-02-01): Fixes #196
fixes #196
commit d89adb81 (2018-02-01): Fixes #196