RedApparat/Fotoapparat #204: Configuration  Issue with Java

DvlopRyanSpears, 2018-02-02:
Hi,i'm trying use this lib with Java,
` mFotoapparat = Fotoapparat
                .with(this)
                .into(cameraView_fotoapparat)           // view which will draw the camera preview
                .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
            //    .photoSize()   // we want to have the biggest photo possible
                .photoResolution(new Function1<Iterable<Resolution>, Resolution>() {
                    @Override
                    public Resolution invoke(Iterable<Resolution> resolutions) {
                        return new Resolution(1080,1200);
                    }
                }).lensPosition(new Function1<Iterable<? extends LensPosition>, LensPosition>() {
                    @Override
                    public LensPosition invoke(Iterable<? extends LensPosition> lensPositions) {
                        return null;  //Here's the issue  that i'm facing. I don't know how to create a LensPosition object. since the LensPosition class is a 'sealed' class and it's child class are not defined as static class inside of LensPosition .
                    }
                })
                // we want to have the biggest photo possible
                .build();`

Diolor, 2018-02-02:
e.g. `LensPositionSelectorsKt.front()`for front camera or `back()` for the back one.

Note that you resolution is also hardcoded which might not work for all devices: some devices might not support 1080x1200 resolution.

closed 2018-02-02 by Diolor

DvlopRyanSpears, 2018-02-02:
Thanks for your reply.
For the resolution problem, Shall i query the device's max supported resolution before i settle ?

Diolor, 2018-02-02:
We already have an example here about highest resolution: https://github.com/Fotoapparat/Fotoapparat/wiki/Configuration-Java

Available selectors are listed here:https://github.com/Fotoapparat/Fotoapparat/tree/master/fotoapparat/src/main/java/io/fotoapparat/selector

ARSOFT1, 2018-10-15:
LensPositionSelectorsKt.front() is not working.....Update your code

ARSOFT1, 2018-10-15:
builder =Fotoapparat.with(this).into(binding.cameraView)
                .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                .photoResolution(ResolutionSelectorsKt.highestResolution())// we want to have the biggest photo possible
                .lensPosition(LensPositionSelectorsKt.front())
                .focusMode(SelectorsKt.firstAvailable(  // (optional) use the first focus mode which is supported by device
                        FocusModeSelectorsKt.continuousFocusPicture(),
                        FocusModeSelectorsKt.autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                        FocusModeSelectorsKt.fixed()             // if even auto focus is not available - fixed focus mode will be used
                ))
                .flash(SelectorsKt.firstAvailable(      // (optional) similar to how it is done for focus mode, this time for flash
                        FlashSelectorsKt.autoRedEye(),
                        FlashSelectorsKt.autoFlash(),
                        FlashSelectorsKt.torch()
                ));