RedApparat/Fotoapparat #205: Laggy preview on debug app builds

chirayudesai, 2018-02-07:
Hi,

I'm trying to integrate Fotoapparat into an app and have noticed that the preview is really laggy on debug builds. On release builds it's fine.
Tried using the same minify/proguard configuration for the debug build as the release build but that didn't help.

#### What are you trying to achieve or the steps to reproduce?

Lag - free preview

#### How did you initialize FA?

2.2.0:
```
Fotoapparat
                .with(this)
                .into(cameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .previewResolution(highestResolution())
                .lensPosition(back())
                .focusMode(firstAvailable(
                        continuousFocusPicture(),
                        autoFocus(),
                        fixed()
                ))
                .frameProcessor(new ScanFrameProcessor(this))
                .build();
```

1.4.1/1.5.0:
```
Fotoapparat
                .with(this)
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .photoSize(biggestSize())
                .lensPosition(back())
                .focusMode(firstAvailable(
                        FocusModeSelectors.continuousFocus(),
                        autoFocus(),
                        fixed()
                ))
                .frameProcessor(new ScanFrameProcessor(this))
                .build();
```

Setting the previewResolution to lowestResolution() doesn't make it lag anymore on debug builds but that's basically unusable, can't read codes with that.

#### Context:

* *FA version*: 1.4.1, 1.5.0, 2.2.0
* *Devices/APIs affected*: Google Pixel XL
* *any other relevant information*:

Example apps:
https://github.com/freeotp/freeotp-android (1.4.1, 1.5.0 would be one line change)
https://github.com/chirayudesai/freeotp-android/tree/fotoapparat-update (above updated to 2.2.0)

chirayudesai, 2018-02-07:
So I'm unsure as to what causes this. It happens on all 3 FA versions I tested, and only on *debug* builds. release builds work just fine, no lag.

Diolor, 2018-02-07:
I really don't believe there is any bug here as there are no reasonable grounds for this to happen.

I expect that the logcat is causing the lag. Can you please disconnect the device from your computer, restart the app and use the debug flavor? Is it still laggy?

chirayudesai, 2018-02-07:
@Diolor Yes this doesn't really sound like a bug with the lib directly as such.

And it can't be logcat, I do have it running for the release builds too and there's not much different in the amount of logging. I did a quick check anyway, doesn't make a difference.