RedApparat/Fotoapparat #207: How to debug "Operation failed internally"?

jaroslavas, 2018-02-13:
#### What are you trying to achieve or the steps to reproduce?

When running `result.saveToFile(image)` photo is not saved and the only thing I get is `Couldn't deliver pending result: Operation failed internally` in logcat and null unit in `whenDone`. How should I debug the issue and find out what's wrong?

#### How did you initialize FA?

```java
fotoapparat = Fotoapparat
                .with(this)
                .into(cameraView)           // view which will draw the camera preview
                .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                .photoResolution(highestResolution())   // we want to have the biggest photo possible
                .lensPosition(back())       // we want back camera
                .focusMode(firstAvailable(  // (optional) use the first focus mode which is supported by device
                        continuousFocusPicture(),
                        autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                        fixed()             // if even auto focus is not available - fixed focus mode will be used
                ))
                .flash(firstAvailable(      // (optional) similar to how it is done for focus mode, this time for flash
                        autoFlash(),
                        torch()
                ))
                .logger(loggers(            // (optional) we want to log camera events in 2 places at once
                        logcat()//,           // ... in logcat
                ))
                .build();
```

And then:

```java
        File image = new File(filePath.getPath());
        PhotoResult result = fotoapparat.takePicture();
        PendingResult<Unit> pending = result.saveToFile(image);
        pending.whenDone(new WhenDoneListener<Unit>() {
            @Override
            public void whenDone(Unit unit) {
                setResult(unit != null ? RESULT_OK : RESULT_CANCELED);
                finish();
            }
        });
```
#### What was the result you received?

`Couldn't deliver pending result: Operation failed internally`

#### What did you expect?

A photo saved to a file or more descriptive error/exception.

#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: Samsung S4 Android 7

Raenar4k, 2018-02-15:
I'm also interested in this - trying to figure out why on specific device app freezes after taking a photo, while only having some logs to look at.
In 1.5.0 version, when camera failed i could get some specifics about exception in log:
```
E/Camera: Error 100
W/System.err: java.lang.Exception
W/System.err:     at io.fotoapparat.hardware.v1.Camera1.recordMethod(Camera1.java:434)
W/System.err:     at io.fotoapparat.hardware.v1.Camera1.takePicture(Camera1.java:285)
```
Now, at 2.2.0 at same device with same issue i see:
```
Couldn't deliver pending result: Operation failed internally.
```
I think it can be useful if `PendingResult. whenAvailable` after catching exception would also print out stacktrace alongside that short description. It might not be _that_ useful in some cases, but still it is better to at least give some idea why that might happen.

Diolor, 2018-02-19:
Hey, we suppress the exception in the `whenDone` (maybe we can think a way to deliver that one somehow). If you really need to debug this you can either use `.await` which will throw an exception or attach an `rxAdapter`  which will propagate the error in the `.onError`

valterh4ck3r, 2018-02-24:
Same problem.

amiyo-biswas, 2019-08-23:
Same issue.