RedApparat/Fotoapparat #210: Photo's are always square

tomtheunit, 2018-02-19:
Whenever I take a photo in portrait mode, the photo always comes out square.
I want it to keep the same aspect ratio as the preview.

```java
private void takePicture() {
        swapView();

        photoResult = fotoapparatSwitcher.getCurrentFotoapparat().takePicture();
        currentPhoto = new File(getExternalFilesDir("photos"), photoEndpoint() + ".jpg");
        photoResult.saveToFile(currentPhoto);

        photoResult
                .toBitmap(scaled(0.25f))
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {
                        capturedPhoto = findViewById(R.id.result);
                        capturedPhoto.setImageBitmap(result.bitmap);
                        capturedPhoto.setRotation(-result.rotationDegrees);
                    }
                });
    }
```

My image view:

```xml
<ImageView
        android:id="@+id/result"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:scaleType="centerCrop"
        android:adjustViewBounds="false"/>
```

#### How did you initialize FA?

```java
private Fotoapparat createFotoapparat(LensPosition position) {
        return Fotoapparat
                .with(this)
                .cameraProvider(CameraProviders.v1()) // change this to v2 to test Camera2 API
                .into(cameraView)
                .previewScaleType(ScaleType.CENTER_CROP)
                .photoSize(standardRatio(biggestSize()))
                .lensPosition(lensPosition(position))
                .focusMode(firstAvailable(
                        continuousFocus(),
                        autoFocus(),
                        fixed()
                ))
                .flash(firstAvailable(
                        autoRedEye(),
                        autoFlash(),
                        torch(),
                        off()
                ))
                .previewFpsRange(rangeWithHighestFps())
                .sensorSensitivity(highestSensorSensitivity())
                .frameProcessor(new SampleFrameProcessor())
                .logger(loggers(
                        logcat(),
                        fileLogger(this)
                ))
                .cameraErrorCallback(new CameraErrorCallback() {
                    @Override
                    public void onError(CameraException e) {
                        Toast.makeText(CameraActivity3.this, e.toString(), Toast.LENGTH_LONG).show();
                    }
                })
                .build();
    }
```

using version 1.4.1

Diolor, 2018-02-19:
Hey! We don't support versions prior to v2. But still if the photos are square then the device only supports square resolutions. Secondly if you use `CENTER_CROP` this will create only a visual crop (or not if you use CENTER_INSIDE).

closed 2018-02-19 by Diolor

tomtheunit, 2018-02-28:
What do you mean the device only supports square resolutions. Surely this is a bug with the framework since the phone is capable of taking 16:9 photos in other apps.

Diolor, 2018-02-28:
I changed nothing related to it in last version but glad it works for you

tomtheunit, 2018-02-28:
Actually I think the issue was with the attributes I had on the preview. I was trying to make it full screen, however it was not covering the whole screen (resulting in a square view). Do you know what dimensions the photo output is by default?

Do you know why this is happening?
[image]

`<ImageView
                android:id="@+id/result"
                android:layout_width="match_parent"
                android:layout_height="match_parent"
                android:adjustViewBounds="true"
                tools:ignore="ContentDescription,RtlHardcoded" />`