RedApparat/Fotoapparat #211: io.fotoapparat.exception.camera.UnsupportedConfigurationException: FocusMode configuration selector couldn't select a value. Supported parameters:

aviontechnology, 2018-02-19:
E/Fotoapparat Example: Camera error: 
                       io.fotoapparat.exception.camera.UnsupportedConfigurationException: FocusMode configuration selector couldn't select a value. Supported parameters: [io.fotoapparat.parameter.FocusMode$Fixed@d4b4b9d]
                           at io.fotoapparat.parameter.camera.provide.CameraParametersProviderKt.getCameraParameters(CameraParametersProvider.kt:158)
                           at io.fotoapparat.hardware.Device.getCameraParameters$suspendImpl(Device.kt:145)
                           at io.fotoapparat.hardware.Device.getCameraParameters(Device.kt)
                           at io.fotoapparat.routine.camera.UpdateConfigurationRoutineKt$updateCameraConfiguration$1.doResume(UpdateConfigurationRoutine.kt:26)
                           at kotlin.coroutines.experimental.jvm.internal.CoroutineImpl.resume(CoroutineImpl.kt:54)
                           at kotlinx.coroutines.experimental.DispatchTask.run(CoroutineDispatcher.kt:129)
                           at kotlinx.coroutines.experimental.EventLoopBase.processNextEvent(EventLoop.kt:147)
                           at kotlinx.coroutines.experimental.BlockingCoroutine.joinBlocking(Builders.kt:236)
                           at kotlinx.coroutines.experimental.BuildersKt.runBlocking(Builders.kt:174)
                           at kotlinx.coroutines.experimental.BuildersKt.runBlocking$default(Builders.kt:167)
                           at io.fotoapparat.routine.camera.UpdateConfigurationRoutineKt.updateCameraConfiguration(UpdateConfigurationRoutine.kt:25)
                           at io.fotoapparat.routine.camera.StartRoutineKt.start(StartRoutine.kt:44)
                           at io.fotoapparat.routine.camera.StartRoutineKt.bootStart(StartRoutine.kt:26)
                           at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:84)
                           at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:34)
                           at io.fotoapparat.concurrent.CameraExecutor$execute$future$1.call(CameraExecutor.kt:26)
                           at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                           at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                           at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                           at java.lang.Thread.run(Thread.java:818)
D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl

Diolor, 2018-02-19:
Your camera supports only `Fixed` focus mode but you don't have it in the list of selectors. Check `firstAvailable` in an example [here](https://github.com/Fotoapparat/Fotoapparat/wiki/Configuration-Kotlin)

closed 2018-02-19 by Diolor

aviontechnology, 2018-02-20:
Hello 
as per suggestion i use below code:
https://github.com/Fotoapparat/Fotoapparat/wiki/Configuration-Kotlin
but 
it start continues crashing.
please tell me what to do ?

Diolor, 2018-02-20:
You need to initialise your lateinit property