RedApparat/Fotoapparat #212: A resource was acquired at attached stack trace but never released.

mmBs, 2018-02-19:
#### What are you trying to achieve or the steps to reproduce?

StrictMode throws an error: ` resource was acquired at attached stack trace but never released. See java.io.Closeable for information on avoiding resource leaks.`

I used the sample app from this repository, added an App class which extends `Application` class and implemented `onCreate()` method where I initialised StrictMode:

```java
class App : Application() {

    override fun onCreate() {

        StrictMode.setThreadPolicy(
            StrictMode.ThreadPolicy.Builder()
                .detectDiskReads()
                .detectDiskWrites()
                .detectNetwork()
                .penaltyLog()
                .build()
        )
        StrictMode.setVmPolicy(
            StrictMode.VmPolicy.Builder()
                .detectLeakedSqlLiteObjects()
                .detectLeakedClosableObjects()
                .penaltyLog()
                .penaltyDeath()
                .build()
        )

        super.onCreate()
    }

```

AndroidManifest.xml
```java
<?xml version="1.0" encoding="utf-8"?>
<manifest package="io.fotoapparat.sample"
          xmlns:android="http://schemas.android.com/apk/res/android">

    <application
        android:allowBackup="true"
        android:name=".App"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">
        <activity
            android:name=".MainActivity"
            android:screenOrientation="fullSensor">
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>

                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
    </application>
```

Steps to reproduce:
1. Run the app.
2. Click back button.
3. Wait 4 secs. In the logcat StrictMode throws an error. 

#### How did you initialize FA?

FA was initialised like in the sample app (actually, the sample app was used to reproduce this issue)

#### What was the result you received?

    E/StrictMode: A resource was acquired at attached stack trace but never released. See java.io.Closeable for information on avoiding resource leaks.
                                                                   java.lang.Throwable: Explicit termination method 'release' not called
                                                                       at dalvik.system.CloseGuard.open(CloseGuard.java:223)
                                                                       at android.view.Surface.setNativeObjectLocked(Surface.java:566)
                                                                       at android.view.Surface.<init>(Surface.java:199)
                                                                       at io.fotoapparat.hardware.CameraDeviceKt.setDisplaySurface(CameraDevice.kt:402)
                                                                       at io.fotoapparat.hardware.CameraDeviceKt.access$setDisplaySurface(CameraDevice.kt:1)
                                                                       at io.fotoapparat.hardware.CameraDevice.setDisplaySurface(CameraDevice.kt:267)
                                                                       at io.fotoapparat.routine.camera.StartRoutineKt.start(StartRoutine.kt:75)
                                                                       at io.fotoapparat.routine.camera.StartRoutineKt.bootStart(StartRoutine.kt:26)
                                                                       at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:84)
                                                                       at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:34)
                                                                       at io.fotoapparat.concurrent.CameraExecutor$execute$future$1.call(CameraExecutor.kt:26)
                                                                       at java.util.concurrent.FutureTask.run(FutureTask.java:266)
                                                                       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
                                                                       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
                                                                       at java.lang.Thread.run(Thread.java:764)

#### What did you expect?

There is no issues coming from StrictMode.

#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: Samsung Galaxy S8+
* *any other relevant information*:

iamutkarshtiwari, 2018-02-20:
Facing the same issue

radosdesign, 2018-02-21:
Same issue here

jgrossophoff, 2018-03-28:
Having the same issue on a Samsaung Galaxy Tab Active T365. I made sure that fa.stop() is called when finishing the activity.

radosdesign, 2018-04-03:
It looks to me that we need to 'release' surface in CameraDevice.kt in stop() method. Something like this:

```
 /**
     * Closes the connection to a camera.
     */
    open fun close() {
        logger.recordMethod()
        surface.release()
        camera.release()
    }
```

Diolor, 2018-07-05:
Fixed in the finally released v2.3.1

closed 2018-07-05 by Diolor