RedApparat/Fotoapparat #214: How to enable 'tap to focus' in my Fotoapparat java activity?

iamutkarshtiwari, 2018-02-19:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

I checked the sample app but the MainActivity.kt is in kotlin and I didn't find any code for adding the tap-to-focus in the Activity.java. Could anyone please point me in the right direction?

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
// Wrap code in markdown source tags
```

#### What was the result you received?

#### What did you expect?

#### Context:

* *FA version*:
* *Devices/APIs affected*:
* *any other relevant information*:

Diolor, 2018-02-19:
Hey! @Diederikjh already helped us with this and added the example in the ActivityJava. Take a look again in the master branch!

closed 2018-02-19 by Diolor

iamutkarshtiwari, 2018-02-20:
@Diolor I am unable to access the ```.focusView()``` method of Fotoapparat builder.

[image]

Diolor, 2018-02-20:
Which version are you using?

iamutkarshtiwari, 2018-02-20:
@Diolor I am using ```2.2.0```. Thank you for the suggestion :)