RedApparat/Fotoapparat #215: Camera preview goes blank when ViewPager pauses and resumes from another fragment

iamutkarshtiwari, 2018-02-21:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

To describe best, here is the issue that I am facing - https://drive.google.com/file/d/1T4IbE9plnR_5fXnoTjFkAwGY7lKHblh6/view?usp=sharing

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
@Nullable
    @Override
    public View onCreateView(LayoutInflater inflater, @Nullable ViewGroup container, @Nullable Bundle savedInstanceState) {
        binding = DataBindingUtil.inflate(inflater, R.layout.fragment_comment_image_picker_camera_tab, null, false);

        return binding.getRoot();
    }

    @Override
    public void onViewCreated(View view, @Nullable Bundle savedInstanceState) {
        super.onViewCreated(view, savedInstanceState);

        if (!isEnabled()) {
            return;
        }

        mCameraView = binding.viewFinder;
        mFocusView = binding.focusView;
        mImageProcessingProgressBar = binding.imageProcessingProgressLoader;

        // Disable buttons until permissions check
        toggleCameraOptionsClickability(false);
        toggleShutterButtonAvailability(false);

        mCameraConfiguration = getCameraConfiguration();
        mFotoapparat = createFotoapparat();
    }

   

    private void setupUi() {
        setupShutterClick();
        setupSwitchCameraClick();
        setupFlashClick();

        binding.infoButton.setOnClickListener(v -> {
            new CameraPhotoPickerTipsDialog().show(getFragmentManager(), TAG_TIPS_DIALOG);
        });
    }

    private Fotoapparat createFotoapparat() {
        mActiveLensPosition = LensPositions.initialMode();
        mActiveFlashMode = FlashModes.initialMode();
        return Fotoapparat
                .with(getContext())
                .into(mCameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .lensPosition(getLensSelector(mActiveLensPosition))
                .flash(getFlashSelector(mActiveFlashMode))
                .cameraErrorCallback(this)
                .build();
    }

    private CameraConfiguration getCameraConfiguration() {
        return CameraConfiguration
                .builder()
                .photoResolution(standardRatio(
                        highestResolution()
                ))
                .focusMode(firstAvailable(
                        continuousFocusPicture(),
                        autoFocus(),
                        fixed()
                ))
                .flash(firstAvailable(
                        autoFlash(),
                        torch(),
                        off()
                ))
                .previewFpsRange(highestFps())
                .sensorSensitivity(highestSensorSensitivity())
                .build();
    }

    @Override
    public void onResume() {
        super.onResume();
        mFotoapparat.start();
        if (mSavePictureSubscription == null || mSavePictureSubscription.isUnsubscribed()) {
            toggleCameraOptionsClickability(true);
            toggleCameraOptionsAvailability(true);
            toggleShutterButtonAvailability(true);
            toggleProgressLoaderLayer(false);
        }
    }

    @Override
    public void onPause() {
        super.onPause();
        mFotoapparat.stop();
        if (mSavePictureSubscription != null && !mSavePictureSubscription.isUnsubscribed()) {
            mSavePictureSubscription.unsubscribe();
        }
    }

    @Override
    public void onStop() {
        super.onStop();
    }

    @Override
    public void onDestroy() {
        super.onDestroy();
    }

```

#### What was the result you received?

#### What did you expect?

#### Context:

* *FA version*:  2.2.0
* *Devices/APIs affected*: Xiomi Redmi Note3 (MIUI9 Android 6.0 Marshmallow)
* *any other relevant information*:

Diolor, 2018-02-21:
Can you please attach a logger an paste here **only the logs with the Fotoapparat** tag?

iamutkarshtiwari, 2018-02-22:
``` 02-22 12:56:11.844 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:56:22.404 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:56:22.906 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:56:32.288 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:56:46.412 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:56:46.870 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:56:52.222 26421-26429/com.kouzoh.mercari.debug E/StrictMode: A resource was acquired at attached stack trace but never released. See java.io.Closeable for information on avoiding resource leaks.
                                                                      java.lang.Throwable: Explicit termination method 'release' not called
                                                                          at dalvik.system.CloseGuard.open(CloseGuard.java:180)
                                                                          at android.view.Surface.setNativeObjectLocked(Surface.java:459)
                                                                          at android.view.Surface.<init>(Surface.java:145)
                                                                          at io.fotoapparat.hardware.CameraDeviceKt.setDisplaySurface(CameraDevice.kt:398)
                                                                          at io.fotoapparat.hardware.CameraDeviceKt.access$setDisplaySurface(CameraDevice.kt:1)
                                                                          at io.fotoapparat.hardware.CameraDevice.setDisplaySurface(CameraDevice.kt:265)
                                                                          at io.fotoapparat.routine.camera.StartRoutineKt.start(StartRoutine.kt:75)
                                                                          at io.fotoapparat.routine.camera.StartRoutineKt.bootStart(StartRoutine.kt:26)
                                                                          at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:84)
                                                                          at io.fotoapparat.Fotoapparat$start$1.invoke(Fotoapparat.kt:34)
                                                                          at io.fotoapparat.concurrent.CameraExecutor$execute$future$1.call(CameraExecutor.kt:26)
                                                                          at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                          at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                          at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                          at java.lang.Thread.run(Thread.java:818)
02-22 12:56:53.087 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:56:53.204 26421-27218/com.kouzoh.mercari.debug E/BufferQueueProducer: [SurfaceTexture-0-26421-1] connect(P): BufferQueue has been abandoned
02-22 12:56:59.841 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:00.270 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:57:00.802 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:57:10.542 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:11.016 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:57:11.555 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:57:13.327 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:13.792 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:57:16.095 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:57:16.212 26421-27218/com.kouzoh.mercari.debug E/BufferQueueProducer: [SurfaceTexture-0-26421-4] connect(P): BufferQueue has been abandoned
02-22 12:57:19.629 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:20.066 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:57:20.421 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:57:22.329 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:22.808 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop
02-22 12:57:23.766 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On resume
02-22 12:57:23.865 26421-27218/com.kouzoh.mercari.debug E/BufferQueueProducer: [SurfaceTexture-0-26421-6] connect(P): BufferQueue has been abandoned
02-22 12:57:24.829 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On pause
02-22 12:57:25.284 26421-26421/com.kouzoh.mercari.debug E/Mercari_JP: On stop```

Diolor, 2018-02-22:
Attach `.logger(logcat())` to your FA builder. We need the logs emitted from this. They have "Fotoapparat" as TAG (instead of `Mercari_JP`)

iamutkarshtiwari, 2018-02-22:
@Diolor Here is the gist for all the Fotoapparat logs for the steps as I did in the video - https://gist.github.com/iamutkarshtiwari/a0063915812e66f76ed575c3f5017cd7

Diolor, 2018-02-22:
Thank you! I see a `Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed` which looks like the issue.

This is happening only if you have the fragment paused, then pause/resume the app. 

I suspect the `CameraView` is still active and being held from your viewpager and when you try to restart the camera it's conflicting the current start with the previous stop.

I think it's an issue with fragments but maybe we have missed something here. 
I need to make a demo to get a hands on understanding

Diolor, 2018-03-22:
Closing this due to inactivity. Feel free to open new issue

closed 2018-03-22 by Diolor

amadeu01, 2018-07-09:
Have any update on that issue?

amadeu01, 2018-07-11:
I did the code below before creating the Fotoapprat instance.

```java
        CameraManager manager =
                (CameraManager) view.getContext().getSystemService(CAMERA_SERVICE);
        try {
            for (String cameraId : manager.getCameraIdList()) {
                CameraCharacteristics chars
                        = manager.getCameraCharacteristics(cameraId);
                App.logException(Log.DEBUG, "PhotoAction", chars.toString());

                manager.openCamera(cameraId, new CameraDevice.StateCallback() {
                    @Override
                    public void onOpened(@NonNull CameraDevice camera) {
                        App.logException(Log.DEBUG, "PhotoAction", "onOpened");
                    }

                    @Override
                    public void onDisconnected(@NonNull CameraDevice camera) {
                        App.logException(Log.DEBUG, "PhotoAction", "onOpened");
                    }

                    @Override
                    public void onError(@NonNull CameraDevice camera, int error) {
                        App.logException(Log.DEBUG, "PhotoAction", "onError");
                        camera.close();
                    }
                }, new Handler());
            }
        } catch (Exception e) {
            e.printStackTrace();
        }

```

It worked. Then, I released that `onCreate` of the Activity was being called twice as well. But, only in one specific device. What I believe be the problem is the number of connections on the camera hardware exceed its limit because of unkilled instance of `CameraView` as @Diolor told.