RedApparat/Fotoapparat #216: Error calling `takePicture` on FA 2.2.0

Archinamon, 2018-02-22:
#### What are you trying to achieve or the steps to reproduce?

Performing takePicture procedure and it fails on some devices repeatedly on version 2.2.0, but works well on 1.5.0.

```java
PhotoResult photoResult = fotoapparat.takePicture();
//....
result.saveToFile(destination).await(); //inside android.os.Handler in another thread
```

#### How did you initialize FA?

```java
Fotoapparat
                .with(context)
                .into(preview)
                .jpegQuality(DEFAULT_QUALITY_PERCENT)
                .photoResolution(getSelectorFunctionBySizeSelector(context, sizeSelector))
                .previewResolution(getSelectorFunctionBySizeSelector(context, sizeSelector))
                .previewScaleType(ScaleType.CenterCrop)
                .focusMode(getDefaultFocusModesSequence())
                .flash(firstAvailable(autoRedEye(), autoFlash(), off()))
                .lensPosition(position.getConductedLensPosition())
                .cameraErrorCallback(errorCallback)
                .build();
```

#### What was the result you received?
Silent exception on 
```anm: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
	at aod.a(FileWriterSaveProcessor.java:23)
	at aoh.a(StatelessReporter.java:89)
	at aoh.a(StatelessReporter.java:14)
	at aoh$2.run(StatelessReporter.java:67)
	at android.os.Handler.handleCallback(Handler.java:820)
	at android.os.Handler.dispatchMessage(Handler.java:104)
	at android.os.Looper.loop(Looper.java:207)
	at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
	at java.util.concurrent.FutureTask.report(FutureTask.java:94)
	at java.util.concurrent.FutureTask.get(FutureTask.java:164)
	at aav.a(PendingResult.kt:52)
	at aod.a(FileWriterSaveProcessor.java:19)
	... 7 more
Caused by: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
	at java.util.concurrent.FutureTask.report(FutureTask.java:94)
	at java.util.concurrent.FutureTask.get(FutureTask.java:164)
	at aav$b.call(PendingResult.kt:33)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
	at java.lang.Thread.run(Thread.java:818)
Caused by: java.lang.RuntimeException: takePicture failed
	at android.hardware.Camera.native_takePicture(Native Method)
	at android.hardware.Camera.takePicture(Camera.java:1892)
	at zg.b(CameraDevice.kt:370)
	at zg.a(CameraDevice.kt:1)
	at zf.e(CameraDevice.kt:141)
	at abk$a.doResume(TakePhotoRoutine.kt:15)
	at aeo.resume(CoroutineImpl.kt:54)
	at akf.run(CoroutineDispatcher.kt:129)
	at akl.d_(EventLoop.kt:147)
	at ajg.i(Builders.kt:236)
	at aji.a(Builders.kt:174)
	at aji.a(Builders.kt:167)
	at abk.a(TakePhotoRoutine.kt:12)
	at yg$f.invoke(Fotoapparat.kt:117)
	at yg$f.invoke(Fotoapparat.kt:34)
	at yo$c.call(CameraExecutor.kt:26)
	... 4 more```

#### What did you expect?

Normally saving image to file.

#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: HUAWEI CRO-U00 (6.0 23 api), HUAWEI TIT-L01 (5.1 22 api), MEIZU M3S (5.1 22 api), LGE LG-K100 (6.0 23 api), 
* *any other relevant information*:

Diolor, 2018-02-22:
This might happen for several reasons e.g. if you request multiple takePicture() in short time. Do you have any guards for that?

dmitry-zaitsev, 2018-02-23:
@Diolor I believe the library is handling sequential `takePicture` requests correctly as it will queue the requests. At least it is supposed to.

Diolor, 2018-02-23:
That's true indeed. Still the Android camera can always throw a RuntimeException. Not sure if we can do something more here or if a specific reproducable case

Archinamon, 2018-02-23:
> This might happen for several reasons e.g. if you request multiple takePicture() in short time. Do you have any guards for that?

Surely I make only one request to `takePicture` and then block the button until I get the 'success' status in callback. But, the thing I'm wonder about — I see two permanent errors leading single `takePicture` call. Everything about Handler and asynchronous handling on picture saving are the same in the decorater around the FA and works well on FA 1.5.0

Maybe some other points I can look for?