RedApparat/Fotoapparat #218: Image is being saved in a different orientation from that of the screen

Nkurayijahubert, 2018-03-03:
Hello,

Am facing an issue, where image is being in different orientation from that of screen, if screen is in portrait mode image will be in landscape and vise-versa. Any help!

Diolor, 2018-03-04:
Front or back camera? Also which device?

scifinder, 2018-03-04:
A similar problem. In landscape mode, the second shot has an incorrect orientation. Steps:
1. Turn the screen horizontally (Samsung Galaxy A3).
2. Take a picture, display it on the preview.
3. Go back. Make a second photo.
4. The second picture will be shot perpendicularly.
[image]

Diolor, 2018-03-04:
That's a different issue @scifinder, thanks. It's already reported in #193

Nkurayijahubert, 2018-03-05:
@Diolor it's on back camera, on Nexus 5x phone.

ShkurtiA, 2018-03-19:
any solutions or workarounds yet?

Nkurayijahubert, 2018-03-24:
I would like to ask the same, @Diolor

Nkurayijahubert, 2018-03-24:
@Diolor,  Any solution?

Diolor, 2018-03-24:
No, but you can send a PR @Nkurayijahubert !

dmitry-zaitsev, 2018-03-26:
@Nkurayijahubert could you please share a sample photo with us?

Nkurayijahubert, 2018-03-26:
[image]

@dmitry-zaitsev Here is the sample image

Nkurayijahubert, 2018-03-26:
@dmitry-zaitsev as you can see phone is in portrait orientation but the image comes in landscape orientation

dmitry-zaitsev, 2018-03-26:
@Nkurayijahubert please share the actual JPG image, not just the screenshot. I am trying to figure out what Exif properties the file has.

Nkurayijahubert, 2018-03-27:
Ok, here it's @dmitry-zaitsev 

[image]

Nkurayijahubert, 2018-03-27:
@dmitry-zaitsev  before to open it's in landscape but in preview it become portrait, as you can see in that comment above.

dmitry-zaitsev, 2018-03-27:
@Nkurayijahubert this is not the bug of Fotoapparat but the missing feature in your image viewer. The JPG file contains information about the image orientation and it is a responsibility of the viewer to rotate the image accordingly.

closed 2018-03-27 by dmitry-zaitsev

Nkurayijahubert, 2018-03-27:
@dmitry-zaitsev what do you mean by image viewer? is it manufacturing issue or how the user has configured?

dmitry-zaitsev, 2018-03-27:
@Nkurayijahubert by the viewer I mean the Android application you use to view the image on your device.

Nkurayijahubert, 2018-03-27:
Ok, Thank you @dmitry-zaitsev let try to fix the issue

antoineFrau, 2019-07-25:
Hi so I was facing this issue, and figure out that I just needed to rotate the image of 90 degrees if the width < height, meaning that the image is in landscape mode.

```kotlin
photoResult?.toBitmap()?.whenAvailable { photo ->
    photo?.apply {
        // Avoid the landscape image preview
       var bitmap = if (this.bitmap.width < this.bitmap.height) this.bitmap else this.bitmap.rotate(90)
        Log.i(LOGGING_TAG, "New photo captured. Bitmap length: ${this.bitmap.byteCount}")
    }
        ?: Log.e(LOGGING_TAG, "Couldn't capture photo.")
}
```