RedApparat/Fotoapparat #220: performClick was lost =)

scifinder, 2018-03-10:
Hello! =)
I decided to implement the zoom with gestures. On my construction `cameraView.setOnTouchListener(new View.OnTouchListener() { ...` I receive a warning: 

> Custom view "CameraView" has setOnTouchListener called on it but does not override performClick.

I tried to extend the class CameraView so to add `@Override performClick(...)`, but I get error like "**final** class can not be overridden". Can you add it on you source? =)

Diolor, 2018-03-10:
We would be happy to accept a PR!

scifinder, 2018-03-10:
I'm not sure, but I think something needs to be done in CameraView class:
```
@Override
public boolean performClick() {
        super.performClick();
        return true;
}
```

dmitry-zaitsev, 2018-03-26:
@scifinder you can safely ignore this warning - it does not really point at any real issues in the implementation. Thanks for reporting though 👍

closed 2018-03-26 by dmitry-zaitsev