RedApparat/Fotoapparat #227: White balance

savviestapps, 2018-03-14:
Sorry if Im way out of line here, but I can't find a discussion group for this.

#### What are you trying to achieve?

I'd like to use this library and set White Balance in the preview and saved image.  I'd like to be able to set custom white balance as well as the enumerated options available in the Camera2 library.  Im using Java language

I was hoping to find something in the CameraConfiguration class but no luck.  Is there a way to get access to the underlying camera/surface objects to set white balance somehow?

Thanks

Diolor, 2018-03-14:
Hey! This feature doesn't exist. But if you like send a PR!
Pattern is quite similar for all attrs.

Docs for camera 1:

https://developer.android.com/reference/android/hardware/Camera.Parameters.html#getWhiteBalance()

savviestapps, 2018-03-14:
Sorry, whats a PR?  Does this mean I can somehow get hold of a Camera object from a Fotoapparat or cameraView object and go to work on it?

dmitry-zaitsev, 2018-03-26:
@savviestapps PR stands for Pull Request. @Diolor was proposing you to fork the GitHub repository and make all the changes yourself. If you are not confident about doing that - no worries, we will get to this feature as soon as we'll have time.

esafirm, 2018-10-16:
Hey @Diolor @dmitry-zaitsev can i take this? 
So i think i just have to add white balance option to `CameraConfiguration`?