RedApparat/Fotoapparat #228: Stair-step effect and noticable edges in FA 2.0 on Nexus 5x, Pixel

Raenar4k, 2018-03-16:
#### What are you trying to achieve or the steps to reproduce?

There are noticeable stair-steps in preview at version 2.2 of FA.
I've managed to reproduce it only on Nexus 5x and Google Pixel 1 gen, not sure if other devices are affected.

I've attached a video with comparison of FA 1.4 and FA 2.2 sample apps.
(Gif big enough too see details was about 100mb, so i've uploaded zip with mp4 here)
[sharp_edges.mp4.zip](https://github.com/Fotoapparat/Fotoapparat/files/1818665/sharp_edges.mp4.zip)

#### How did you initialize FA?
It is sample app from FA version 1.4 and version 2.2

#### Context:
* *Devices/APIs affected*: Nexus 5x (8.0.0), Google Pixel (8.1.0)

Diolor, 2018-03-16:
What's your `previewResolution` in pixels?

Raenar4k, 2018-03-16:
It is 1600x1200 (1600 width, 1200 height) in both FA 1.4 and 2.2 for Nexus 5x
Im using fotoapparat sample apps from these commits:
9a8bbb10214acabdd3b60db21e6a093045d38986 - 2.2
21652d836e783c99e417465353366c14a52ff769 - 1.4

dmitry-zaitsev, 2018-03-26:
@Raenar4k just an assumption, but it might be related to the size of the camera viewport. As you can see in v1 we had a toolbar which reduced the size of the `CameraView` which in turn possibly affected the image resolution. 

Can you try to reduce the size of the `CameraView` and tell us if there is any noticeable difference?

dmitry-zaitsev, 2018-04-17:
I am closing this until more information is provided.

closed 2018-04-17 by dmitry-zaitsev

Raenar4k, 2018-06-19:
@dmitry-zaitsev Sorry for long reply, finally got to test this.
I've tried same style with toolbar as in 1.4.1 for sample project - and effect is still there.
In my app layout was not changed when migrating to 2.0, so i'm sure that was not the reason.

Also i've not seen this in other devices for now.