RedApparat/Fotoapparat #230: Take picture, nothing happens

BorisLeMeec, 2018-03-21:
#### What are you trying to achieve or the steps to reproduce?

Hi, when I try to take a picture with Fotoapparat, nothing happens.

```java
fun seeAccount(view: View) {
        Log.e("photo", "take")
        val photoResult = fotoapparat.takePicture()
        val byteArrayOutputStream = ByteArrayOutputStream()

        photoResult.toBitmap().whenAvailable { result ->
            Log.i("photo", result.toString())
    }
```

#### How did you initialize FA?

In a good way i think because I am able to see preview in CameraView

#### What was the result you received?

Nothing in Logcat

#### What did you expect?

Something like "Photo : xxxxx:

#### Context:

* * FA version 2.2.0:
* *Xiaomi Mi Mix*:

cyberrob, 2019-07-26:
In my case it's due to invalid file path. After fixing the path issue, camera works as before.

covenant3717, 2021-02-12:
> #### What are you trying to achieve or the steps to reproduce?
> Hi, when I try to take a picture with Fotoapparat, nothing happens.
> 
> ```java
> fun seeAccount(view: View) {
>         Log.e("photo", "take")
>         val photoResult = fotoapparat.takePicture()
>         val byteArrayOutputStream = ByteArrayOutputStream()
> 
>         photoResult.toBitmap().whenAvailable { result ->
>             Log.i("photo", result.toString())
>     }
> ```
> 
> #### How did you initialize FA?
> In a good way i think because I am able to see preview in CameraView
> 
> #### What was the result you received?
> "photo : take"
> Then Nothing
> 
> #### What did you expect?
> Something like "Photo : take
> Photo : xxxxx"
> 
> #### Context:
> * * FA version 2.2.0:
> * _Xiaomi Mi Mix_:

Through experimentation, I found out that the problem is precisely in "toBitmap".
On my device, converting to Bitmap was throwing "java.lang.OutOfMemoryError: Failed to allocate a 63701004 byte allocation with 16777216 free bytes and 42MB until OOM".
After adding (android: largeHeap = "true") to Manifest in <application ... /> - "toBitmap" started working