RedApparat/Fotoapparat #231: jpegQuality doesn't seem to work

ashishkrishnan, 2018-03-22:
I'm trying to compress the image taken by camera using the `jpegQuality` method, but it doesn't seem to work. Attached are the code snippets I used to achieve quality reduction.

```
    fotoapparat = Fotoapparat
        .with(cameraView.context)
        .into(cameraView)
        .previewScaleType(ScaleType.CenterCrop)
        .lensPosition(back())
        .jpegQuality(manualJpegQuality(70))
        .focusMode(firstAvailable(continuousFocusPicture(), autoFocus(), fixed()))
        .logger(logcat())
        .build()
```

I also tried setting the quality using a `CameraConfiguration`. Unfortunately, this also doesn't work.
```
fotoapparat.updateConfiguration(CameraConfiguration(jpegQuality = manualJpegQuality(70)))
```

* Fotoapparat version: 2.2.0
* Tested for API(s): 7.0.1

Diolor, 2018-03-22:
What means "work" ? What do you get, and what are you expectations?

ashishkrishnan, 2018-03-22:
Trying to output a lesser quality image.

Diolor, 2018-03-22:
Well this is what the android camera gives. We cannot do anything about it. Out of curiosity can you try `manualJpegQuality(10)` ?

ashishkrishnan, 2018-03-22:
:stuck_out_tongue:  Did that already. No change. It is the same result. 

Like you said, if that is what the camera gives what is the use of `jpeqQuality` method in the Fotoapparat.Builder?

What does it exactly do?

ashishkrishnan, 2018-03-22:
Hahah, the functionality, I guessed it correctly. But, talking of documentation looks like you referenced a Camera v1 API. 

Does the Camera2 API also support this?. I couldn't find any API exposure in  the corresponding documentation. 

My Testing device Lenovo P2 running Nougat qualifies for Camera2. 

Will check with other devices as well.

Diolor, 2018-03-22:
Fotoapparat uses camera 1. You can try use directly camera 2 instead of Fotoapparat if it fits better your needs

closed 2018-03-22 by Diolor

ashishkrishnan, 2018-03-22:
Oh, I didn't realise that. Do you plan to write a wrapper over the Camera2 API or in your opinion is it needed?

Diolor, 2018-03-22:
We had. Unfortunately camera 2 is not as performant as camera 1, so we dropped it. Camera 1 can work with all devices. Also the ones 21+

ashishkrishnan, 2018-03-22:
Well, that's true. Although, My Team and I used the `toBitmap`() to collect and the bitmap and then reduce using `Bitmap.compress(......)` and then pass it down. 

Hence, we look good with this. Thank you for the clarifications.

ashishkrishnan, 2018-03-23:
@Diolor Hey, I might have found something. 
setting a jpegQuality does work. It gives me smaller compressed images. 

Although, rotating the flash by updating the camera configuration actually resets the jpeg quality settings. 

Here are the logs: 
`Before setting the custom flash config
New camera parameters are: CameraParameters
                                   flashMode:		io.fotoapparat.parameter.Flash$Off@53a0673
                                   focusMode:		io.fotoapparat.parameter.FocusMode$ContinuousFocusPicture@37dd0eb
                                   jpegQuality:		50
                                   previewFpsRange:		FpsRange(min=15000, max=30000)
                                   antiBandingMode:		io.fotoapparat.parameter.AntiBandingMode$Auto@23276f4
                                   sensorSensitivity:		null
                                   pictureResolution:		Resolution(width=4128, height=3096)
                                   previewResolution:		Resolution(width=1440, height=1080)
`

After configuration is updated: 
`Fotoapparat$updateConfiguration$1: invoke
D: Device: updateConfiguration
D: CameraDevice: getCapabilities$suspendImpl
D: CameraDevice: updateParameters$suspendImpl
New camera parameters are: CameraParameters
                                   flashMode:		io.fotoapparat.parameter.Flash$Off@53a0673
                                   focusMode:		io.fotoapparat.parameter.FocusMode$ContinuousFocusPicture@37dd0eb
                                   jpegQuality:		90`

GurpreetSK95, 2018-03-26:
@ashishkrishnan @Diolor facing the same issue in my application as well. Probable cause might be the default quality value of 90 which is set in CameraConfiguration. It resets itself everytime the camera configuration is updated.

Diolor, 2018-03-26:
@GurpreetSK95 Are you using `CameraConfiguration` inside the `updateConfiguration`? You should use `UpdateConfiguration` instead

ashishkrishnan, 2018-03-26:
```
  val configuration = UpdateConfiguration
        .builder()
        .flash { toFlashSelector(flash) }
        .jpegQuality { 55 }
        .build()
    fotoapparat.updateConfiguration(configuration)
  }
```

Still the same.

Diolor, 2018-03-26:
The method has no lifecycle. If you mean the `cameraConfiguration: CameraConfiguration` object then it's the same as a FA instance or until changed

455908068, 2019-06-25:
I think the problem is that in the “Device. kt” class, the code snippet is as follows

internal fun updateConfiguration(
        savedConfiguration: CameraConfiguration,
        newConfiguration: Configuration
) = CameraConfiguration(
        flashMode = newConfiguration.flashMode ?: savedConfiguration.flashMode,
        focusMode = newConfiguration.focusMode ?: savedConfiguration.focusMode,
        exposureCompensation = newConfiguration.exposureCompensation
                ?: savedConfiguration.exposureCompensation,
        frameProcessor = newConfiguration.frameProcessor ?: savedConfiguration.frameProcessor,
        previewFpsRange = newConfiguration.previewFpsRange ?: savedConfiguration.previewFpsRange,
        sensorSensitivity = newConfiguration.sensorSensitivity
                ?: savedConfiguration.sensorSensitivity,
        pictureResolution = newConfiguration.pictureResolution
                ?: savedConfiguration.pictureResolution,
        previewResolution = newConfiguration.previewResolution
                ?: savedConfiguration.previewResolution
)

The attribute reset for jpegQuality is missing.

455908068, 2019-06-25:
I think the problem is "Device. Kt "class, code snippet as follows
`
/**
 * Updates the device's configuration.
 */
internal fun updateConfiguration(
        savedConfiguration: CameraConfiguration,
        newConfiguration: Configuration
) = CameraConfiguration(
        flashMode = newConfiguration.flashMode ?: savedConfiguration.flashMode,
        focusMode = newConfiguration.focusMode ?: savedConfiguration.focusMode,
        exposureCompensation = newConfiguration.exposureCompensation
                ?: savedConfiguration.exposureCompensation,
        frameProcessor = newConfiguration.frameProcessor ?: savedConfiguration.frameProcessor,
        previewFpsRange = newConfiguration.previewFpsRange ?: savedConfiguration.previewFpsRange,
        sensorSensitivity = newConfiguration.sensorSensitivity
                ?: savedConfiguration.sensorSensitivity,
        pictureResolution = newConfiguration.pictureResolution
                ?: savedConfiguration.pictureResolution,
        previewResolution = newConfiguration.previewResolution
                ?: savedConfiguration.previewResolution
)`
The attribute reset for jpegQuality is missing.