RedApparat/Fotoapparat #232: Camera takes much time to get a photo

SergeyTrukhachev, 2018-03-24:
Hi guys,

First of all, I'd like to thank for such a great library. It helps me a lot.

Regarding my question, I've noticed that Fotoapparat takes much time to get a photo especially comparing to CameraKit library in `still` mode. 
Are there any ways to speed up this process? I've played with both resolutions (preview and picture) and tried to set up different settings, but capture time stays pretty much the same. 
If you can share some ideas here, I'd really appreciate this.

Thanks.

#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: Galaxy S8+

SergeyTrukhachev, 2018-03-24:
Looks like it is duplicate of #100, so recommendations stay the same?

Diolor, 2018-03-24:
Hi. Depends if you have the same configuration as in #100 . How do you capture and read that photo?

Diolor, 2018-07-04:
Will close this for inactivity

closed 2018-07-04 by Diolor