RedApparat/Fotoapparat #234: Photo file is 0 bytes in saveToFile().whenDone

akirmse, 2018-03-27:
I recently upgraded from version 1.5 to 2.2.  My project is in Java and everything had been working fine for months.  After the upgrade and the requisite syntax changes (because the library is now Kotlin), I am observing that in saveToFile().whenDone(), the file size of the saved photo is 0 bytes, as if it hasn't been written yet.

```
               PhotoResult photoResult = mFotoapparat.takePicture();
                photoResult.saveToFile(photoFile).whenDone((Unit unit) -> {
                    // Did we get a picture?
```

I had been calling whenAvailable() but I can't get that to compile in 2.2.  Do I have to change anything else?

This is my initialization code

```
        mFotoapparat = Fotoapparat
                .with(getActivity())
                .into(cameraView)           // view that will draw the camera preview
                .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                .photoResolution(highestResolution())   // we want to have the biggest photo possible
                .build();
        mFotoapparat.start();
```

scifinder, 2018-03-27:
Hi! See my question: https://github.com/Fotoapparat/Fotoapparat/issues/198

akirmse, 2018-03-28:
Yes, thanks, I am seeing the problem after I made similar changes in upgrading to 2.2.

akirmse, 2018-04-01:
Any thoughts on this?  I've had to revert to 1.5.

fyzs-n9001, 2018-04-15:
The truth is that it's useless to call 'whenDone' or 'whenAvailable' when you want to saveToFile.

akirmse, 2018-04-16:
Can you say more about why it's useless?  It looks like this function was
added explicitly for the same use case I have, namely, knowing when writing
to a file has completed and the file is ready to be used by other code.

See https://github.com/Fotoapparat/Fotoapparat/issues/24

akirmse, 2019-01-26:
The problem was on my side.  I was calling stop() before whenDone() completed.

closed 2019-01-26 by akirmse