RedApparat/Fotoapparat #235: Java support

AviranAbady, 2018-03-30:
Still using Fotoapparat 1.x
Exploring the idea to upgrade to 2.x, getting some issues with upgrading the code (while remaining in Java)

Not sure how the following legacy code should be handled in 2.x:

## Focus listener:
```
fotoApparat.focus().whenAvailable(new PendingResult.Callback<FocusResult>() {
            @Override
            public void onResult(FocusResult focusResult) {
                if (focusResult == FocusResult.Focused) {
               
                } else if (focusResult == FocusResult.UnableToFocus) {
                
                }
            }
        });
```

Pending result seem to have changed, not sure how to apply the current Kotlin callback machanism while still writing in `Java`.
`FocusResult` enum is now an object type, should `instanceof` be used instead as a compare operator?

## Taking a picture

```
fotoApparat.takePicture().toPendingResult().whenAvailable(new PendingResult.Callback<Photo>() {
                @Override
                public void onResult(Photo photo) {

                }

});```   

What's the proper way to handle the async operation / supplying the callback?

Please provide code examples.

AviranAbady, 2018-04-01:
@Diolor @dmitry-zaitsev can you assist?

AroundPixels, 2018-04-10:
@AviranAbady you can use Kotlin version with your Java code, no problem!

AviranAbady, 2018-04-15:
@AroundPixels can you provide a code example on how to provide a Kotlin 'listener' in a java class?
In this case for example
```java

fotoApparat.takePicture().toPendingResult().whenAvailable(new PendingResult.Callback<Photo>() {
                @Override
                public void onResult(Photo photo) {

                }

});
```

AviranAbady, 2018-04-16:
Providing a code example, for everyone not familiar with Kotlin. Basicly Kotlin code can be integrated directly in the Java code, the callbacks should be defined as lambdas and return Unit.INSTANCE.

**build.gradle**
```
android {
   ...
   ...
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}
```
 
```java
 fotoApparat.takePicture().
                    toPendingResult().
                    whenAvailable(photo -> {
                       /* handle photo */
                        return Unit.INSTANCE;
                    });

fotoApparat.focus().whenAvailable(focusResult -> {
              
            return Unit.INSTANCE;
        });
```

closed 2018-04-16 by AviranAbady